ansys/pyaedt #5213: Vector graphic export of 2D geometry and mesh (svg format)

mheroth, 2024-09-26:
### Description of the feature

Export of Mesh and of the 2D geometry as scalable vector graphic (svg).

SMoraisAnsys, 2024-09-27:
Hi @mheroth, could you add a bit more information to the feature you would like to see in `pyaedt` ?

mheroth, 2024-10-07:
I would like to be able to export the 2D geometry (modeled in Electronics Desktop Maxwell 2D) directly as an .svg file. Similar to how it is already possible as a .step file, for example. In addition to the 2D geometry, it would be important for me to be able to export the mesh used. So that in the end the 2D geometry and the mesh are described using corner points and connection types.