ansys/pyaedt #5243: Error while Exporting Field Results Using PyAEDT

chamanth-vct, 2024-10-04:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hi,

I am working on a project where I need to export field results, and I have automated this workflow using PyAEDT. However, I encountered the following error during the export process:

`PyAEDT ERROR: case file format is not supported for this plot.`

The export works perfectly through the GUI, but the issue occurs when I try to automate it using PyAEDT.

### Steps To Reproduce

I was able to reproduce the issue with the following code and the sample aedt file ([link to download](https://apps.vcollab.com/fileshare/downloadsharedfile?file_ID=3ob5k4m193cbmz)).

**Code:**

```py
import pyaedt

file_path = r"C:\Users\chamanth\testing_error\Project3.aedt"
exports_folder_path = r"C:\Users\chamanth\testing_error\exports_folder"

desktop = pyaedt.Desktop(non_graphical=True) 
prj= desktop.load_project(file_path)
model = prj.modeler
part_names = model.object_names

post = prj.post  # post_processor

#Exporting Result

for name in part_names:

    h_field_plot = post.create_fieldplot_volume([name],quantity="Vector_H",plot_name="H_vector_plot")

    export_status = post.export_field_plot(plot_name = "H_vector_plot",
                            output_dir = exports_folder_path,
                            file_name = name,
                            file_format = 'case'
                                        )

   #if export_status:
   #   print(f"Exported H_Vector for {name}")
   # else:
   #    print(f"Not Exported for {name}")

    h_field_plot.delete()

desktop.close_desktop()

```

In the above code, I automate the export of the H vector field for each object (part). The aedt file contains 10 parts in total. The code successfully exports data for 4 parts, but for the remaining 6, the error is thrown as shown below:

**Error Log:**
```txt
..
PyAEDT INFO: Modeler class has been initialized! Elapsed time: 0m 0sec
PyAEDT INFO: Parsing C:/Users/chamanth/Downloads/testing_error/Project3.aedt.
PyAEDT INFO: File C:/Users/chamanth/Downloads/testing_error/Project3.aedt correctly loaded. Elapsed time: 0m 0sec
PyAEDT INFO: aedt file load time 0.1261458396911621
PyAEDT INFO: PostProcessor class has been initialized! Elapsed time: 0m 0sec
PyAEDT INFO: Post class has been initialized! Elapsed time: 0m 0sec
PyAEDT ERROR: case file format is not supported for this plot.
PyAEDT ERROR: case file format is not supported for this plot.
PyAEDT ERROR: case file format is not supported for this plot.
PyAEDT ERROR: case file format is not supported for this plot.
PyAEDT ERROR: case file format is not supported for this plot.
PyAEDT ERROR: case file format is not supported for this plot.
PyAEDT INFO: Desktop has been released and closed.
```

**Additional Information**
- The same export process works fine through the GUI for the parts where automation fails.
- I encountered this issue with large number of aedt files.
- I initially raised this issue in a discussion (#5182 ). However, based on feedback from the discussion, it seems this might be a bug in PyAEDT, hence I’m reporting it here.

Could anyone assist in resolving this issue? Any suggestions or insights would be highly appreciated.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
asttokens==2.4.1
attrs==24.2.0
certifi==2024.8.30
cffi==1.17.1
charset-normalizer==3.3.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.2
contourpy==1.3.0
cycler==0.12.1
debugpy==1.8.5
decorator==5.1.1
defusedxml==0.7.1
exceptiongroup==1.2.2
executing==2.1.0
fonttools==4.53.1
fpdf2==2.7.9
idna==3.10
ipykernel==6.29.5
ipython==8.27.0
jedi==0.19.1
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
jupyter_client==8.6.3
jupyter_core==5.7.2
kiwisolver==1.4.7
matplotlib==3.9.2
matplotlib-inline==0.1.7
nest-asyncio==1.6.0
numpy==1.26.4
packaging==24.1
pandas==2.2.2
parso==0.8.4
pillow==10.4.0
platformdirs==4.3.6
plumbum==1.8.3
pooch==1.8.2
prompt_toolkit==3.0.47
psutil==6.0.0
pure_eval==0.2.3
pyaedt==0.10.2
pycparser==2.22
pydantic==2.9.2
pydantic_core==2.23.4
pyedb==0.28.0
Pygments==2.18.0
pyparsing==3.1.4
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.2
pyvista==0.44.1
pywin32==306
PyYAML==6.0.2
pyzmq==26.2.0
referencing==0.35.1
requests==2.32.3
rpds-py==0.20.0
rpyc==6.0.0
Rtree==1.3.0
scikit-rf==1.3.0
scipy==1.14.1
scooby==0.10.0
six==1.16.0
stack-data==0.6.3
toml==0.10.2
tornado==6.4.1
traitlets==5.14.3
typing_extensions==4.12.2
tzdata==2024.1
urllib3==2.2.3
vtk==9.3.1
wcwidth==0.2.13

nathmay, 2024-10-04:
Hi @chamanth-vct,

I had a quick look at your project. Some of your parts are conductors, and so by default "solve_inside" for them is off. When you create a volume plot, those return empty plots and throw error during export.

In the AEDT GUI, it automatically switches to a surface plot when you select such an object, so you still get a meaningful plot there. This behavior is not present in the current PyAEDT version.

You can still achieve this by manually checking. Try the following:

```python
for name in part_names:

    # Get the Object3D object associated with current name.
    obj =  model.get_object_from_name(name)

    # Check if the object has "solve_inside" set. If yes, create a volume plot.
    #  If not, create a surface plot on its faces.
    if obj.solve_inside:
        h_field_plot = post.create_fieldplot_volume([name],quantity="Vector_H",plot_name="H_vector_plot")
    else:
        h_field_plot = post.create_fieldplot_surface(obj.faces,quantity="Vector_H",plot_name="H_vector_plot")

    export_status = post.export_field_plot(plot_name = "H_vector_plot",
                            output_dir = exports_folder_path,
                            file_name = name,
                            file_format = 'case'
                                        )

    h_field_plot.delete()
```

gmalinve, 2024-10-07:
Hi @chamanth-vct,

@nathmay is right. I just implemented a check on the create_fieldplot_volume for the solve_inside property (#5256).
If you are trying to plot in an object where solve_inside is False, this will issue a warning and return false.

Thanks for feedback.

Regards,

Giulia

chamanth-vct, 2024-10-08:
Hi @nathmay  and @gmalinve

Thank you for your suggestion.

I tried implementing your recommendation to check for `solve_inside`. Unfortunately, I’m still encountering the same error. It seems that the `create_field_surface()` function is not generating the plots in the same way that `create_field_volume()` does. 

Here's the code I used:
```py
import pyaedt

file_path = r"C:\Users\chamanth\Downloads\testing_error\Project3.aedt"
exports_folder_path = r"C:\Users\chamanth\Downloads\testing_error\exports_folder_new"

desktop = pyaedt.Desktop() 
prj= desktop.load_project(file_path)
model = prj.modeler
part_names = model.object_names
post = prj.post  # post_processor

for name in part_names:
    obj =  model.get_object_from_name(name)

    if obj.solve_inside:
        h_field_plot = post.create_fieldplot_volume([name],quantity="Vector_H",plot_name="H_vector_plot")
    else:
        h_field_plot = post.create_fieldplot_surface(obj.faces,quantity="Vector_H",plot_name="H_vector_plot")

    export_status = post.export_field_plot(plot_name = "H_vector_plot",
                            output_dir = exports_folder_path,
                            file_name = name,
                            file_format = 'case')

    h_field_plot.delete()

desktop.close_desktop()
```

And here's the error message:

```txt
PyAEDT INFO: Parsing C:/Users/chamanth/Downloads/testing_error/Project3.aedt.
PyAEDT INFO: File C:/Users/chamanth/Downloads/testing_error/Project3.aedt correctly loaded. Elapsed time: 0m 0sec
PyAEDT INFO: aedt file load time 0.17937850952148438
PyAEDT INFO: PostProcessor class has been initialized! Elapsed time: 0m 0sec
PyAEDT INFO: Post class has been initialized! Elapsed time: 0m 0sec
PyAEDT INFO: Parsing design objects. This operation can take time
PyAEDT INFO: 3D Modeler objects parsed. Elapsed time: 0m 0sec
PyAEDT ERROR: case file format is not supported for this plot.
PyAEDT ERROR: case file format is not supported for this plot.
PyAEDT ERROR: case file format is not supported for this plot.
PyAEDT ERROR: case file format is not supported for this plot.
PyAEDT ERROR: case file format is not supported for this plot.
PyAEDT ERROR: case file format is not supported for this plot.
PyAEDT INFO: Desktop has been released and closed.
```

Could you please take another look and let me know if there’s anything else I might be missing?

Thank you again for your help!

Regards
Chamanth

gmalinve, 2024-10-09:
Hi @chamanth-vct ,

when you try to create the surface plot please provide the intrinsics. By default you can get them from the setup:
`h_field_plot = hfss.post.create_fieldplot_surface(obj.faces, quantity="Vector_H", plot_name="H_vector_plot", intrinsics=hfss.setups[0].default_intrinsics)`.
In the same PR I'm implementing an enhancement that if intrinsics are not provided the default ones are automatically taken.

Also bear in mind that pyaedt has become an alias of ansys.aedt.core in the import statements. Please review this page: [breaking changes](https://aedt.docs.pyansys.com/version/stable/release_1_0.html#release-1-0).

Hope this helps,

Regards,

Giulia

closed 2024-10-09 by Samuelopez-ansys