ansys/pyaedt #5258: ERROR with AEDT is not installed on your system when running unitest

anspraksaph, 2024-10-07:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When using PyAedt With this commit hash: 99b3187517e463d15dfdfef1cf6491c56945ddeb, unable to run test C:\pyaedt\_unittest_solvers\test_26_emit.py with this error "AEDT is not installed on your system. Install AEDT version 2022 R2 or higher."

There is no problem with this commit hash: c9df21df4435f4e0b966727ec5d8cc24f8d984db (When using with the same AEDT version)(I'm using EDT 25.1)

### Steps To Reproduce

1. Clone PyAedt at "C:/pyaedt". 
2. With Python 3.11, set virtual environment, activate it and run this command:
- python -m pip install pip -U
- python -m pip install wheel setuptools -U
- python -m pip install C:/pyaedt\.[tests]
- python -m pip install pytest-azurepipelines
- pytest -v C:\pyaedt\_unittest_solvers\test_26_emit.py

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.11

### Installed packages

(test_env_3.10) C:\Users\Public\Desktop\PyAedt_Terminal>python -m pip freeze
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
asttokens==2.4.1
attrs==24.2.0
certifi==2024.8.30
cffi==1.17.1
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.3.0
coverage==7.6.1
cycler==0.12.1
decorator==5.1.1
defusedxml==0.7.1
et-xmlfile==1.1.0
exceptiongroup==1.2.2
execnet==2.1.1
executing==2.1.0
fiona==1.10.1
fonttools==4.54.1
fpdf2==2.8.1
geopandas==0.14.4
idna==3.10
imageio==2.35.1
importlib_resources==6.4.5
iniconfig==2.0.0
ipython==8.27.0
jedi==0.19.1
joblib==1.4.2
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
kiwisolver==1.4.7
markdown-it-py==3.0.0
matplotlib==3.9.2
matplotlib-inline==0.1.7
mdurl==0.1.2
meshio==5.3.5
mock==5.1.0
networkx==3.3
numpy==1.26.4
openpyxl==3.1.5
osmnx==1.9.4
packaging==24.1
pandas==2.2.3
parso==0.8.4
pillow==10.4.0
platformdirs==4.3.6
pluggy==1.5.0
plumbum==1.9.0
pooch==1.8.2
prompt_toolkit==3.0.48
psutil==6.0.0
pure_eval==0.2.3
pyaedt @ file:///C:/pyaedt
pycparser==2.22
pydantic==2.9.2
pydantic_core==2.23.4
pyedb==0.29.0
Pygments==2.18.0
pyparsing==3.1.4
pyproj==3.7.0
pytest==8.3.3
pytest-azurepipelines==1.0.5
pytest-cov==5.0.0
pytest-nunit==1.0.7
pytest-xdist==3.6.1
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.2
pyvista==0.44.1
pywin32==307
PyYAML==6.0.2
referencing==0.35.1
requests==2.32.3
rich==13.9.2
rpds-py==0.20.0
rpyc==6.0.1
Rtree==1.3.0
scikit-learn==1.5.2
scikit-rf==1.3.0
scipy==1.14.1
scooby==0.10.0
shapely==2.0.6
six==1.16.0
SRTM.py==0.3.7
stack-data==0.6.3
threadpoolctl==3.5.0
toml==0.10.2
tomli==2.0.2
traitlets==5.14.3
typing_extensions==4.12.2
tzdata==2024.2
urllib3==2.2.3
utm==0.7.0
vtk==9.3.1
wcwidth==0.2.13

SMoraisAnsys, 2024-10-08:
Hi @anspraksaph and thanks for the bug report.
Can you give a try at the changes of #5264 to see if that solves your problem ?

anspraksaph, 2024-10-08:
Hi @SMoraisAnsys  with the change of https://github.com/ansys/pyaedt/pull/5264, it's working okay. Thank you

SMoraisAnsys, 2024-10-14:
Closing following #5264

closed 2024-10-14 by SMoraisAnsys