ansys/pyaedt #5263: Icepak. Bug in manual mesh setting with icepak.mesh.global_mesh_setting

kinielsen, 2024-10-08:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hello Team, 

The manual mesh settings in Icepak has stopped working. 
We get the error below directly from AEDT:

PyAEDT INFO: Mesh class has been initialized! Elapsed time: 0m 0sec
PyAEDT ERROR: Setting not available.

Can you please help?

I am using Python version 3.12.0 and pyaedt version 0.10.3

Best regards, 
Kim 

### Steps To Reproduce

ipk = Icepak(project="TEST",non_graphical=False)

ipk.save_project()

glob_msh = ipk.mesh.global_mesh_region

glob_msh.manual_settings = True
glob_msh.settings["MaxElementSizeX"] = "2mm"
glob_msh.settings["MaxElementSizeY"] = "3mm"
glob_msh.settings["MaxElementSizeZ"] = "4mm"
glob_msh.settings["MaxSizeRatio"] = 2
glob_msh.settings["UserSpecifiedSettings"] = True
glob_msh.settings["UniformMeshParametersType"] = "XYZ Max Sizes"
glob_msh.settings["MaxLevels"] = 2
glob_msh.settings["BufferLayers"] = 1
glob_msh.update()

time.sleep(30)

ipk.save_project()

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.11

### Installed packages

annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
attrs==24.2.0
bcrypt==4.2.0
blinker==1.8.2
certifi==2024.8.30
cffi==1.17.1
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.3.0
cryptography==43.0.1
cycler==0.12.1
dash==2.17.1
dash-core-components==2.0.0
dash-design-kit==1.6.7
dash-enterprise-auth==0.2.2
dash-html-components==2.0.0
dash-table==5.0.0
defusedxml==0.7.1
dill==0.3.8
diskcache==5.6.3
et-xmlfile==1.1.0
fast_simplification==0.1.7
fiona==1.10.1
Flask==3.0.3
fonttools==4.54.1
fpdf2==2.7.9
geopandas==0.14.4
idna==3.10
imageio==2.35.1
importlib_metadata==8.5.0
itsdangerous==2.2.0
Jinja2==3.1.4
joblib==1.4.2
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
kiwisolver==1.4.7
markdown-it-py==3.0.0
MarkupSafe==2.1.5
matplotlib==3.9.2
mdurl==0.1.2
meshio==5.3.5
multiprocess==0.70.16
nest-asyncio==1.6.0
networkx==3.3
numpy==1.26.4
openpyxl==3.1.5
osmnx==1.9.4
packaging==24.1
pandas==2.2.3
paramiko==3.5.0
pillow==10.4.0
platformdirs==4.3.6
plotly==5.24.1
plumbum==1.8.3
pooch==1.8.2
psutil==6.0.0
pyaedt==0.10.3
pycparser==2.22
pydantic==2.8.2
pydantic_core==2.20.1
pyedb==0.27.0
Pygments==2.18.0
PyJWT==2.9.0
PyNaCl==1.5.0
pyparsing==3.1.4
pyproj==3.6.1
python-dateutil==2.9.0.post0
pytz==2024.2
pyvista==0.44.1
pywin32==306
PyYAML==6.0.2
referencing==0.35.1
requests==2.32.3
retrying==1.3.4
rich==13.8.1
rpds-py==0.20.0
rpyc==6.0.1
Rtree==1.3.0
scikit-learn==1.5.2
scikit-rf==1.3.0
scipy==1.14.1
scooby==0.10.0
setuptools==75.1.0
shapely==2.0.6
six==1.16.0
SRTM.py==0.3.7
tenacity==9.0.0
threadpoolctl==3.5.0
toml==0.10.2
typing_extensions==4.12.2
tzdata==2024.2
urllib3==2.2.3
utm==0.7.0
vtk==9.3.1
Werkzeug==3.0.4
wheel==0.44.0
zipp==3.20.2

lorenzovecchietti, 2024-10-10:
Hi Kim, thanks for opening the issue.

The problem that I see in your lines is this:
`glob_msh.settings["UserSpecifiedSettings"] = True`

`"UserSpecifiedSettings"` key should not be used, as that property is implicitly set with `glob_msh.manual_settings = True`.

kinielsen, 2024-10-10:
Hi @lorenzovecchietti ,

Thank you for the comments. 
I removed the glob_msh.settings["UserSpecifiedSettings"]  from the code - and get no error.
However, the Global Mesh Settings are not updated, as you see in the picture below:

[image]

I wonder what causes the problem? 

Best regards, 
Kim

lorenzovecchietti, 2024-10-10:
You're right. I'm fixing this

kinielsen, 2024-10-11:
Thank you @lorenzovecchietti !

PR #5275 (2024-10-14): FIX: Select "user specified settings" setting more reliably
Close #5263
closed 2024-10-14 by lorenzovecchietti