ansys/pyaedt #5269: Ansys.aedt.core.hfss Problem

EngAnsys, 2024-10-09:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hello all,

I tried to run some really simple codes (available in the tutorial website of PyAEDT) in the command window of Ansys electronics desktop. However, I keep getting this error.
[image]

### Steps To Reproduce

>>> import ansys.aedt.core
with ansys.aedt.core.Desktop(specified_version="2023.2", non_graphical=True, new_desktop_session=True, close_on_exit=True,
             student_version=False):
    circuit = ansys.aedt.core.Circuit()
ImportException: No module named ansys.aedt.core

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.11

### Installed packages

annotated-types==0.7.0
ansys-additive-core==0.18.2
ansys-api-additive==1.7.2
ansys-api-platform-instancemanagement==1.1.0
ansys-platform-instancemanagement==1.1.2
ansys-pythonnet==3.1.0rc3
attrs==24.2.0
cachetools==5.5.0
certifi==2024.8.30
cffi==1.17.1
charset-normalizer==3.3.2
clr-loader==0.2.6
colorama==0.4.6
contourpy==1.3.0
cycler==0.12.1
defusedxml==0.7.1
dill==0.3.9
fonttools==4.54.1
fpdf2==2.8.1
google-api-core==2.20.0
google-api-python-client==2.148.0
google-auth==2.35.0
google-auth-httplib2==0.2.0
googleapis-common-protos==1.65.0
grpcio==1.66.2
grpcio-health-checking==1.62.3
httplib2==0.22.0
idna==3.10
importlib_metadata==8.5.0
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
kiwisolver==1.4.7
matplotlib==3.9.2
numpy==1.26.4
OpenMagneticsVirtualBuilder==0.5.4
packaging==24.1
pandas==2.2.3
pillow==10.4.0
platformdirs==4.3.6
plumbum==1.9.0
pooch==1.8.2
proto-plus==1.24.0
protobuf==4.25.5
psutil==6.0.0
pyaedt==0.11.0
pyasn1==0.6.1
pyasn1_modules==0.4.1
pycparser==2.22
pydantic==2.9.2
pydantic_core==2.23.4
pyedb==0.29.0
PyMKF==0.9.4
pyparsing==3.1.4
python-dateutil==2.9.0.post0
pytz==2024.2
pyvista==0.44.1
pywin32==307
PyYAML==6.0.2
referencing==0.35.1
requests==2.32.3
rpds-py==0.20.0
rpyc==6.0.1
rsa==4.9
Rtree==1.3.0
scikit-rf==1.3.0
scipy==1.14.1
scooby==0.10.0
six==1.16.0
toml==0.10.2
tqdm==4.66.5
typing_extensions==4.12.2
tzdata==2024.2
uritemplate==4.1.1
urllib3==2.2.3
vtk==9.3.1
zipp==3.20.2

Samuelopez-ansys, 2024-10-09:
Hi @EngAnsys ,

From your screenshot I guess you are using the Ironpython AEDT console. PyAEDT is for CPython coding.

Please follow these steps to install PyAEDT if you do not how to install it manually like any other open source library.

https://aedt.docs.pyansys.com/version/stable/Getting_started/Installation.html#install-from-pyaedt-installer

Regards

closed 2024-10-09 by Samuelopez-ansys

EngAnsys, 2024-10-09:
Hi @Samuelopez-ansys ,

Thank you for your answer. I already have the PyAEDT installed. Exactly from the link you sent me. 
In this case, where should I place the code I showed on the screenshot? (sorry, I'm just starting with this)

Samuelopez-ansys, 2024-10-09:
You opened the native AEDT terminal which is Ironpython, and PyAEDT is not installed there.

You should open the PyAEDT console:
[image]

EngAnsys, 2024-10-09:
Now, it works.

Thank you @Samuelopez-ansys !