ansys/pyaedt #5273: Bug located in Maxwell3d

AlfVII, 2024-10-10:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hi,

with the latest version of pyaedt, when I run a program that opens the Desktop, it closes it once it finishes, even if I specify not to.

### Steps To Reproduce

```
import pyaedt

project = pyaedt.Maxwell3d(
    projectname="Putabidatete",
    solution_type="EddyCurrent",
    non_graphical=False,
    new_desktop_session=True,
    close_on_exit=False
)
```

It opens the GUI, creates a project, and closes the GUI

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.11

### Installed packages

annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
attrs==24.2.0
cffi==1.17.1
clr-loader==0.2.6
defusedxml==0.7.1
fonttools==4.54.1
fpdf2==2.8.1
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
numpy==1.26.4
OpenMagneticsVirtualBuilder==0.5.4
pandas==2.2.3
pillow==10.4.0
plumbum==1.9.0
psutil==6.0.0
pyaedt==0.11.0
pycparser==2.22
pydantic==2.9.2
pydantic_core==2.23.4
pyedb==0.30.0
PyMKF==0.9.11
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.2
pywin32==307
PyYAML==6.0.2
referencing==0.35.1
rpds-py==0.20.0
rpyc==6.0.1
Rtree==1.3.0
scikit-rf==1.3.0
scipy==1.14.1
six==1.16.0
toml==0.10.2
typing_extensions==4.12.2
tzdata==2024.2

Samuelopez-ansys, 2024-10-14:
@AlfVII @SMoraisAnsys Indeed in PyCharm I can reproduce it, but when I try from an Ipython console, it is working:

[image]

The workaround I found in PyCharm is including the code in a "with" context:

```
import ansys.aedt.core

with ansys.aedt.core.Maxwell3d(
    projectname="Putabidatete2",
    solution_type="EddyCurrent",
    non_graphical=False,
    new_desktop_session=True,
    close_on_exit=False,
) as max:

    pass
```

AlfVII, 2024-10-14:
I am getting it directly from running it using Powershell with pyaedt==3.11.0. It I donwgrade to 3.10.0 it works fine.

@Samuelopez-ansys the "with ... as ..." workaround works in my system, but I cannot really use it for what origined this issue (the Ansyas project), because in it I create an class object in the that I reuse along the class:
[image]

closed 2024-10-14 by Samuelopez-ansys