ansys/pyaedt #5282: Add Missing Circuit Elements Needed for ETK

DaveTwyman, 2024-10-11:
### Description of the feature

Migration work of the transformer toolkit from ACT to PyAEDT needs the following elements that are missing from PyAEDT

Add to the MaxwellCircuitComponents class
-Sinusoidal Current Source
-Sinusoidal Voltage Source

Add to the CircuitComponents class
-CreatePage
-SelectPage

jvela018, 2025-02-03:
@DaveTwyman ,

There are two potential work arounds here. 
1. Use the create_component method
2. In case that you want an actual intuitive method with the name of the component you can create a wrapper

Based on the ETK you can create an instance of MaxwellCircuit (although, I'd rather go through Maxwell3D)
```
        app = MaxwellCircuit(design="circuit_" + design_name)
        schematic = app.modeler.schematic
```

Let us create variables for schematics location=[0,0] and page number: 4
```python
page = 4 
x = 0 
y = 0
```

 Case (1) you can create a Voltage source as follows

```python
    component_name = schematic.create_component(
                component_library="Sources",
                component_name="VSin",
                location=[x, y],
                angle=-90,
                page=("Page{}".format(page))
            )
```
in case (2) you could just wrap case (1) so that assigning the voltage source looks like

```
def create_voltage_source(location, page):
    schematic.create_component(
                component_library="Sources",
                component_name="VSin",
                location=[x, y],
                angle=-90,
                page=("Page{}".format(page))
            )
```

We could become even more creative and create a simple source that includes both voltage/current sources 

```
def create_source(type=VSin, location, page):
    schematic.create_component(
                component_library="Sources",
                component_name=type,
                location=[x, y],
                angle=-90,
                page=("Page{}".format(page))
            )
```

Note that this method in addition to PR #5712 , should be enough to handle this issue. Maybe we can reconsider PR #5283

Samuelopez-ansys, 2025-04-27:
@jvela018 could we close this issue?

jvela018, 2025-04-28:
@DaveTwyman is it ok if we close this issue?

DaveTwyman, 2025-04-28:
@jvela018 @Samuelopez-ansys , Yes lets close it

closed 2025-04-28 by DaveTwyman