ansys/pyaedt #5304: Error when opening version 24.1 after version 23.2 in pyaedt

RickKuKu, 2024-10-16:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

There's error message after running the following code.

maxwell = Maxwell3d( version="2023.2" )
maxwell.release_desktop()

maxwell = Maxwell3d( version="2024.1" )
maxwell.release_desktop()
[image]

However, it's ok when I reverse the sequence.

maxwell = Maxwell3d( version="2024.1" )
maxwell.release_desktop()

maxwell = Maxwell3d( version="2023.2" )
maxwell.release_desktop()

### Steps To Reproduce

Run the following code.

from ansys.aedt.core import Maxwell3d

maxwell = Maxwell3d( version="2023.2" )
maxwell.release_desktop()

maxwell = Maxwell3d( version="2024.1" )
maxwell.release_desktop()

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

alabaster==0.7.13
annotated-types==0.7.0
ansys-motorcad-core==0.6.0
ansys-pythonnet==3.1.0rc3
arrow==1.2.3
astroid==2.15.6
asttokens==2.4.0
atomicwrites==1.4.1
attrs==23.1.0
Automat==22.10.0
autopep8==2.0.4
Babel==2.12.1
backcall==0.2.0
bcrypt==4.0.1
beautifulsoup4==4.12.2
binaryornot==0.4.4
black==23.9.1
bleach==6.0.0
certifi==2023.7.22
cffi==1.15.1
chardet==5.2.0
charset-normalizer==3.2.0
click==8.1.7
cloudpickle==2.2.1
clr-loader==0.2.6
colorama==0.4.6
comm==0.1.4
constantly==15.1.0
construct==2.10.68
contourpy==1.1.0
cookiecutter==2.3.0
cryptography==41.0.3
cycler==0.11.0
debugpy==1.8.0
decorator==5.1.1
defusedxml==0.7.1
diff-match-patch==20230430
dill==0.3.7
docstring-to-markdown==0.12
docutils==0.20.1
et-xmlfile==1.1.0
exceptiongroup==1.1.3
executing==1.2.0
fastjsonschema==2.17.1
flake8==6.0.0
fonttools==4.42.1
fpdf2==2.7.9
hyperlink==21.0.0
idna==3.4
imagesize==1.4.1
importlib-metadata==6.8.0
incremental==22.10.0
inflection==0.5.1
intervaltree==3.1.0
ipykernel==6.25.2
ipython==8.15.0
ipython-genutils==0.2.0
isort==5.12.0
jaraco.classes==3.3.0
jedi==0.18.2
jellyfish==1.0.0
Jinja2==3.1.2
joblib==1.3.2
jsonschema==4.19.0
jsonschema-specifications==2023.7.1
jupyter_client==8.3.1
jupyter_core==5.3.1
jupyterlab-pygments==0.2.2
keyring==24.2.0
kiwisolver==1.4.4
lazy-object-proxy==1.9.0
markdown-it-py==3.0.0
MarkupSafe==2.1.3
matplotlib==3.7.2
matplotlib-inline==0.1.6
mccabe==0.7.0
mdurl==0.1.2
mistune==3.0.1
more-itertools==10.1.0
mypy-extensions==1.0.0
nbclient==0.8.0
nbconvert==7.8.0
nbformat==5.9.2
nest-asyncio==1.5.7
numpy==1.25.2
numpydoc==1.5.0
openpyxl==3.1.5
packaging==23.1
pandas==2.0.3
pandocfilters==1.5.0
paramiko==3.3.1
parso==0.8.3
pathspec==0.11.2
pexpect==4.8.0
pickleshare==0.7.5
Pillow==10.0.0
platformdirs==3.10.0
pluggy==1.3.0
plumbum==1.8.2
ply==3.11
pooch==1.8.2
prompt-toolkit==3.0.39
protobuf==4.24.3
psutil==5.9.5
ptyprocess==0.7.0
pure-eval==0.2.2
pyaedt==0.11.0
pyasn1==0.5.0
pycodestyle==2.10.0
pycparser==2.21
pycryptodomex==3.18.0
pydantic==2.8.2
pydantic_core==2.20.1
pydocstyle==6.3.0
pyedb==0.30.0
pyflakes==3.0.1
Pygments==2.16.1
pylint==2.17.5
pylint-venv==3.0.2
pyls-spyder==0.4.0
PyNaCl==1.5.0
pyparsing==3.0.9
PyQt5==5.15.9
PyQt5-Qt5==5.15.2
PyQt5-sip==12.12.2
PyQtWebEngine==5.15.6
PyQtWebEngine-Qt5==5.15.2
pysmi==0.3.4
pysnmp==4.4.12
python-dateutil==2.8.2
python-lsp-black==1.3.0
python-lsp-jsonrpc==1.1.1
python-lsp-server==1.7.4
python-slugify==8.0.1
pythonnet==3.0.2
pytomlpp==1.0.13
pytoolconfig==1.2.5
pytz==2023.3
pyvista==0.44.1
pywin32==306
pywin32-ctypes==0.2.2
PyYAML==6.0.1
pyzmq==25.1.1
QDarkStyle==3.1
qstylizer==0.2.2
QtAwesome==1.2.3
qtconsole==5.4.4
QtPy==2.4.0
referencing==0.30.2
requests==2.31.0
rich==13.5.2
rope==1.9.0
rpds-py==0.10.2
rpyc==6.0.0
Rtree==1.3.0
scikit-learn==1.3.0
scikit-rf==1.3.0
scipy==1.11.2
scooby==0.10.0
six==1.16.0
snowballstemmer==2.2.0
sortedcontainers==2.4.0
soupsieve==2.5
Sphinx==7.2.5
sphinxcontrib-applehelp==1.0.7
sphinxcontrib-devhelp==1.0.5
sphinxcontrib-htmlhelp==2.0.1
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.3
sphinxcontrib-serializinghtml==1.1.9
spyder==5.4.5
spyder-kernels==2.4.4
stack-data==0.6.2
tensor==0.3.6
text-unidecode==1.3
textdistance==4.5.0
threadpoolctl==3.2.0
three-merge==0.1.1
tinycss2==1.2.1
toml==0.10.2
tomli==2.0.1
tomlkit==0.11.8
tornado==6.3.2
traitlets==5.9.0
Twisted==23.8.0
twisted-iocpsupport==1.0.4
typing_extensions==4.7.1
tzdata==2023.3
ujson==5.8.0
urllib3==2.0.3
vtk==9.3.1
watchdog==3.0.0
wcwidth==0.2.6
webencodings==0.5.1
whatthepatch==1.0.5
wrapt==1.15.0
yapf==0.40.1
zipp==3.15.0
zope.interface==6.0

Samuelopez-ansys, 2024-10-16:
@RickKuKu It is weird, could you put a time.sleep() ? Maybe the process is not killed and the OS needs a time to kill everything.

You should even be able to open in the same script multiple desktops using this setting:
settings.use_multi_desktop = True

I would also use GRPC by default:

settings.use_grpc_api = True

RickKuKu, 2024-10-17:
@Samuelopez-ansys  Thanks! It works when I use the setting "settings.use_multi_desktop = True".
If I set "time.sleep(5)" and "settings.use_grpc_api = True" without "settings.use_multi_desktop = True", it can't work.

If I open version 24.1 first and then open version 23.2, it can work even with the setting "settings.use_multi_desktop = **False**".
Is it an expected result?

closed 2025-04-27 by Samuelopez-ansys