ansys/pyaedt #5306: Add "ignore errors" flag to import_materials_from_file

isaacwaldron, 2024-10-16:
### Description of the feature

Customer has requested a flag to allow ignoring errors that occur when importing materials with `import_materials_from_file`, e.g. missing keys that are treated as required.  When successful, the method currently returns a list of the names of the imported materials.  If backwards compatibility is desired, we could retain the existing return signature and expect the user to compare the imported materials against the expected values (i.e. the keys of the JSON's `materials` dictionary).  If not required, we could extend the return signature to return an additional list of materials that could not be imported due to error.

SMoraisAnsys, 2024-10-21:
@isaacwaldron You seem to be pretty up to date with the problem, would you like to work on it ? :D
If not, could you add an example of data provided by a user with missing keys ?

SMoraisAnsys, 2024-11-06:
@isaacwaldron Do you still plan to work on this ?