ansys/pyaedt #5322: Add `edit_source` / `edit_sources` functionality for Eigenmode solutions

nzani, 2024-10-21:
### Description of the feature

Include the ability to edit the source distribution for Eigenmode-type solutions, as is possible in the GUI via Field Overlays -> Edit Sources... option for a simulation with this solution type

### Steps for implementing the feature

Under `hfss.py`, modify `edit_sources` to have particular functionality given the `hfss.solution_type` parameter is "Eigenmode". Specifically, format the `setting` object for the desired eigenmode setup (I would do this myself but I am struggling with finding/guessing the proper format for this).

nzani, 2024-10-21:
Additionally - `edit_sources_from_file` does not work with Eigenmode solution type

Samuelopez-ansys, 2024-10-22:
Hi @nzani ,

Thank you for your feedback, I have fixed the issue: https://github.com/ansys/pyaedt/pull/5330

PR #5330 (2024-10-23): FIX:  Eigenmode Edit source
Close #5322 

Edit sources in Eigenmode was not working.

I also merged functionalities of edit_source/edit_sources and edit_source_from_file, which were similar, and we should not maintain both, so I added a deprecation.
closed 2024-10-23 by Samuelopez-ansys