ansys/pyaedt #5328: Add method to order coil terminals in Transient A-Phi Formulation

AlfVII, 2024-10-22:
### Description of the current limitations

When working with coil terminals and windings in the Transient A-Phi Formulation Ansys requires that you order the coils after adding them. This can be done in the GUI by right-clicking on the winding, and in IronPython by the `oModule.OrderCoilTerminals`. I found no way of doing this with PyAedt.

### Code sample expected

Maybe something like 
`project.order_coil_terminals(winding, list_of_terminals)` ?

nunesanderson, 2025-06-04:
The API was fixed.
@gmalinve just letting you know.

gmalinve, 2025-06-04:
@nunesanderson I don't see any implementation. Where is it?

closed 2025-07-09 by gmalinve