ansys/pyaedt #5329: Bug located in integration with Aedt

AlfVII, 2024-10-22:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I am trying to run a command in Aedt for ordering the coil terminals, as it is not implemented in PyAedt.

When working with Transient A-Phi formulation and windings+coil terminals, Ansys requires that the coil be ordered. In the GUI that is done by right clicking in the winding and selecting Order Coil Terminals
[image]

I want to automate that in Python, so I recorded in IronPython while I did it in the GUI. I ended up with the following code:
```
oModule = oDesign.GetModule("BoundarySetup")
oModule.OrderCoilTerminals(
	[
		"NAME:Primary_winding_parallel_0", 
		"Primary_Parallel_0_Turn_0_copper_terminal", 
		"Primary_Parallel_0_Turn_1_copper_terminal"
	])
```
If I try to run that exact command in Python, I tried:
```
project.get_module("BoundarySetup").OrderCoilTerminals(
[
    "NAME:Primary_winding_parallel_0", 
    "Primary_Parallel_0_Turn_0_copper_terminal", 
    "Primary_Parallel_0_Turn_1_copper_terminal"
])
```
But I get the following error:
```
Traceback (most recent call last):
  File "C:\Users\alfonso.martinez\AppData\Local\Programs\Python\Python311\Lib\site-packages\ansys\aedt\core\generic\grpc_plugin_dll_class.py", line 118, in __Invoke__
    ret = _retry_ntimes(
          ^^^^^^^^^^^^^^
  File "C:\Users\alfonso.martinez\AppData\Local\Programs\Python\Python311\Lib\site-packages\ansys\aedt\core\generic\general_methods.py", line 863, in _retry_ntimes      
    raise AttributeError("Error in Executing Method {}.".format(function.__name__))
AttributeError: Error in Executing Method InvokeAedtObjMethod.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\alfonso.martinez\ansyas\ansyas_mas_loader.py", line 40, in <module>
    ansyas.create_magnetic(
  File "C:\Users\alfonso.martinez\ansyas\ansyas.py", line 344, in create_magnetic
    self.excitation_builder.add_excitation(
  File "C:\Users\alfonso.martinez\ansyas\excitation.py", line 106, in add_excitation
    self.project.get_module("BoundarySetup").OrderCoilTerminals(
  File "C:\Users\alfonso.martinez\AppData\Local\Programs\Python\Python311\Lib\site-packages\ansys\aedt\core\generic\grpc_plugin_dll_class.py", line 138, in DynamicFunc  
    return self.__Invoke__(funcName, args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\alfonso.martinez\AppData\Local\Programs\Python\Python311\Lib\site-packages\ansys\aedt\core\generic\grpc_plugin_dll_class.py", line 129, in __Invoke__   
    raise GrpcApiError("Failed to execute gRPC AEDT command: {}".format(funcName))
ansys.aedt.core.generic.general_methods.GrpcApiError: Failed to execute gRPC AEDT command: OrderCoilTerminals
```

### Steps To Reproduce

Written down in the description

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.11

### Installed packages

annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
attrs==24.2.0
cffi==1.17.1
clr-loader==0.2.6
defusedxml==0.7.1
fonttools==4.54.1
fpdf2==2.8.1
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
numpy==1.26.4
OpenMagneticsVirtualBuilder==0.5.4
pandas==2.2.3
pillow==10.4.0
plumbum==1.9.0
psutil==6.0.0
pyaedt==0.10.0
pycparser==2.22
pydantic==2.9.2
pydantic_core==2.23.4
pyedb==0.30.0
PyMKF @ file:///C:/Users/alfonso.martinez/Downloads/pymkf-0.9.13-cp311-cp311-win_amd64.whl#sha256=7d8cd7c1257a4bd0ca162449886122f4d7d3d5dc2e62801f6bfa18b3fb022e99       
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.2
pywin32==307
PyYAML==6.0.2
referencing==0.35.1
rpds-py==0.20.0
rpyc==6.0.1
Rtree==1.3.0
scikit-rf==1.3.0
scipy==1.14.1
six==1.16.0
toml==0.10.2
typing_extensions==4.12.2
tzdata==2024.2

gmalinve, 2024-10-22:
Hi @AlfVII,

Thanks for opening this issue.
I tried on my side with a dummy file and it works.
Although this is not implemented in PyAEDT, you can access the "BoundarySetup" thanks to oboundary.
In my case it works like that:
`self.aedtapp.oboundary.OrderCoilTerminals(["NAME:Winding_65LIKI","Coil_PTJ5R7", "Coil_ETDM14"])`
Could you also try to get rid of any indentation inside OrderCoilTerminals?

Let me know,

Thanks,

Giulia

AlfVII, 2024-10-22:
Hi Giulia,

I tried removing the indentation, but still fails.

I managed to reproduce it in a small project, attached in a zip. 
[ansyas.zip](https://github.com/user-attachments/files/17474128/ansyas.zip)

I am running Ansys 2024 R1, and when I execute test.py, which load the existing project and tries to order the coil I get the same error as before

gmalinve, 2024-10-22:
Hi Alfonso,

I tested your case with both 2024.1 and 2024.2 and I got a strange behavior.
If I open AEDT without touching the excitations and running the method as is:
`project.oboundary.OrderCoilTerminals(["NAME:Primary_winding_parallel_0", "Primary_Parallel_0_Turn_0_copper_terminal", "Primary_Parallel_0_Turn_1_copper_terminal"])`
I get the same error as you but if I manually change the order of the coils within AEDT and re-run the same line it works.
Still, If I change it back to the original order: 
1) Primary_Parallel_0_Turn_0_copper_terminal
2) Primary_Parallel_0_Turn_1_copper_terminal
and run the same line it works.

Either is an issue with AEDT or the native API in both cases.
Let me investigate further and I'll let you know.
This will be useful anyway for the PyAEDT method implementation.

Kind regards,

Giulia

closed 2024-10-22 by gmalinve