ansys/pyaedt #5366: Axial flux motor in RMxprt

lijinlong0308, 2024-11-01:
HI all,
        is it possible for us to build an axial flux motor with RMxprt using pyaedt? I can not find an example to achieve this.

gmalinve, 2024-11-05:
Hi @lijinlong0308, 

There isn't any example specific to an axial flux motor with PyAEDT but there is an example of an Adjust-Speed Synchronous Machine: [https://examples.aedt.docs.pyansys.com/version/dev/examples/low_frequency/motor/aedt_motor/rmxpert.html](url).
This serves you as an example on how to interact with a Rmxprt session but you can adapt your script based on the machine type you are interested in. You can access the general, stator, rotor properties in the same way.

Hope this is clear.

Kind regards,

Giulia

lijinlong0308, 2024-11-07:
Hi @gmalinve,
     Thanks for you tips.
     I checked the source code, maybe there is no similar modules like ASSM.
      Best,
              Jinlong

lijinlong0308, 2024-11-08:
Hi, @gmalinve ,
      Thanks for your help.
      This issue is solved.
      Best,
                    Jinlong

closed 2024-11-08 by Samuelopez-ansys