ansys/pyaedt #5413: PyAEDT Installer / Installation

FgPy93, 2024-11-12:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

The latest PyAEDT Installer which is provided on the website (https://aedt.docs.pyansys.com/version/stable/Getting_started/Installation.html) does not work properly. 

Didn't figure out if the problem was in the file itself or in the repositories pulled from.

In the end of the installation the preinstalled plug ins (PyAEDT Console, Extension Manager etc.) do not work. 

There are two problems: 
1. The pyaedt_utils.py file which is installed in the Toolkits folder in the Personal Lib is not executable --> In the "new" version of the file f-strings are used. These f-strings cause problems when executing the methods

2. In the "run_....py" file which executes the final python extension file there is one line which add the path where the pyaedt_utils.py file can be found: extensions_dir = os.path.abspath(os.path.join(os.path.dirname(__file__), '..', '..')) --> In the files generated with the new installer the extensions_dir path is wrong. Instead of two levels above the file it is indicated that it can be found on one level above like: extensions_dir = os.path.abspath(os.path.join(os.path.dirname(__file__), '..'))

These faults lead to a not working set up which is not able to call even the PyAEDT Console from AEDT afterwards. 

Substituting the new pyedt_utils.py with the old file where the string manipulations were done with .format() commands and fixing the path location to the correct one in the run_....py files was actually the solution to fix the installation.

### 📝 Steps to reproduce

Download the latest installer and execute like described in AEDT.

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

AEDT 2024R2.

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
Standard configuration which comes with the installer
```

RobPasMue, 2024-11-12:
Transfering issue to PyAEDT repository - pinging @SMoraisAnsys @Samuelopez-ansys

Samuelopez-ansys, 2024-11-12:
Duplicated #5412

closed 2024-11-12 by Samuelopez-ansys