ansys/pyaedt #5428: Add Resistive Sheet boundary condition in Maxwell

IreneWoyna-HE, 2024-11-14:
### Description of the feature

Dear pyAEDT team! 

To me it seems that Resistive Sheet feature is missing in pyAEDT, could you please add it? It's quite similar to insulating boundary condition, the only inputs are the surface which it should be assigned to, and the resistance value.

Cheers
Irene

IreneWoyna-HE, 2024-11-15:
this is the IronPython code:

oModule = oDesign.GetModule("BoundarySetup")
oModule.AssignResistiveSheet(
	[
		"NAME:ResistiveSheet1",
		"Objects:="		, ["contact_inner"],
		"Resistance:="		, "1ohm"
	])

gmalinve, 2024-11-22:
@IreneWoyna-HE FYI I just open a PR for this!

closed 2024-11-26 by gmalinve