ansys/pyaedt #5443: Bug located in export_model_obj

SMoraisAnsys, 2024-11-18:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

Recent changes to `export_model_obj` (see https://github.com/ansys/pyaedt/pull/5432) inverted the logic associated to the export into one or multiple objects. From my understanding, the argument `export_as_single_objects` had the right meaning : "exporting the model into multiple objects" and we should revert the changes that got merged. If the name is confusing then we might want to rename the argument, for example using `export_as_multiple_objects` to export multiple objects or `export_as_single_object` to export into one object.

>NOTE: The changes make the pyaedt-examples repo failing. Specificly, this example https://github.com/ansys/pyaedt-examples/blob/main/examples/electrothermal/components_csv.py

If it is not clear

### Steps To Reproduce

See above

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

Doesn't matter

SMoraisAnsys, 2024-11-18:
Pinging @lorenzovecchietti @Samuelopez-ansys for visibility

lorenzovecchietti, 2024-11-18:
You are absolutely right, sorry. In my mind, I read it without the final "s" and that changed completely the meaning. If you want, I can revert the changes, or you can do it.

SMoraisAnsys, 2024-11-18:
I'll do the changes and modify the name to make it more reader friendly. It is true that the name is a bit miss leading.

closed 2024-11-18 by SMoraisAnsys