ansys/pyaedt #5462: Accessing pyaedt from IronPython Command window

SK-mechcoder, 2024-11-21:
To access the pyaedt from the IronPython command window, I copied the pyaedt folder from the installed location to the this folder
C:\Users\....................\Documents\Ansoft\PersonalLib\pyeaedt
but I get this error message when I try to use pyaedt
[image]

What should I do now ?

RobPasMue, 2024-11-21:
Issue transferred from the PyMAPDL repository. Pinging @ansys/pyaedt-maintainers  for visibility

Samuelopez-ansys, 2024-11-21:
Hi @SK-mechcoder,

PyAEDT dropped the Ironpython compatibility in v0.9.0, it is documented in the Readme, under the PyAEDT compatibility requirements section:
https://github.com/ansys/pyaedt/blob/main/README.md

As suggested by @mikerife, PyAEDT is a CPython library, but in previous versions PyAEDT gave Ironpython support (during 2 years), to easily transfer old Ironpython scripts to CPython.  PyAEDT roadmap is public: https://github.com/orgs/ansys/projects/12

If you want to use PyAEDT and Ironpython, you will need to download PyAEDT 0.8.11 (the latest version compatible with Ironpython) and paste it into the PersonalLib as you did.

I will close the issue because this is not currently supported, and we encourage you to move to CPython, we have created an easy installer to avoid this kind of problem:

https://aedt.docs.pyansys.com/version/stable/Getting_started/Installation.html#install-from-pyaedt-installer

closed 2024-11-21 by Samuelopez-ansys

SK-mechcoder, 2024-11-21:
Thanks for the solution. The doubt raised because of how it is given in the ALH platform.