ansys/pyaedt #5466: Bug located in ... AEDT 2023.2

SK-mechcoder, 2024-11-21:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

So, when I try to create a New desktop using new_desktop=True the Ansys electronics desktop doesn't do the operation as intended,
It just opens without any project or design(attached the image below)
[image]

But when I open Ansys electronics desktop manually and run the below script with new_desktop=False, the code works without any problem(attached the image below)
[image]

### Steps To Reproduce

from ansys.aedt.core import Maxwell3d
m3d=Maxwell3d(project="RT Relay",non_graphical=False,new_desktop=True)
m3d=m3d.rename_design("Iteration 1")

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

annotated-types           0.7.0
ansys-pythonnet           3.1.0rc4
anyio                     4.6.2.post1
argon2-cffi               23.1.0
argon2-cffi-bindings      21.2.0
arrow                     1.3.0
asttokens                 2.4.1
async-lru                 2.0.4
attrs                     24.2.0
babel                     2.16.0
beautifulsoup4            4.12.3
bleach                    6.2.0
certifi                   2024.8.30
cffi                      1.17.1
charset-normalizer        3.4.0
click                     8.1.7
click-plugins             1.1.1
cligj                     0.7.2
cloudpickle               3.1.0
clr-loader                0.2.6
colorama                  0.4.6
comm                      0.2.2
contourpy                 1.3.0
cycler                    0.12.1
debugpy                   1.8.8
decorator                 5.1.1
defusedxml                0.7.1
et_xmlfile                2.0.0
exceptiongroup            1.2.2
executing                 2.1.0
fastjsonschema            2.20.0
fiona                     1.10.1
fonttools                 4.54.1
fpdf2                     2.8.1
fqdn                      1.5.1
geopandas                 0.14.4
h11                       0.14.0
httpcore                  1.0.6
httpx                     0.27.2
idna                      3.10
imageio                   2.36.0
ipycanvas                 0.13.3
ipyevents                 2.0.2
ipykernel                 6.29.5
ipython                   8.29.0
ipython-genutils          0.2.0
ipyvtklink                0.2.3
ipywidgets                7.8.5
isoduration               20.11.0
jedi                      0.19.2
Jinja2                    3.1.4
joblib                    1.4.2
json5                     0.9.28
jsonpointer               3.0.0
jsonschema                4.23.0
jsonschema-specifications 2024.10.1
jupyter_client            8.6.3
jupyter_core              5.7.2
jupyter-events            0.10.0
jupyter-lsp               2.2.5
jupyter_server            2.14.2
jupyter_server_terminals  0.5.3
jupyterlab                4.2.5
jupyterlab_pygments       0.3.0
jupyterlab_server         2.27.3
jupyterlab_widgets        1.1.11
kiwisolver                1.4.7
markdown-it-py            3.0.0
MarkupSafe                3.0.2
matplotlib                3.9.2
matplotlib-inline         0.1.7
mdurl                     0.1.2
meshio                    5.3.5
mistune                   3.0.2
nbclient                  0.10.0
nbconvert                 7.16.4
nbformat                  5.10.4
nest-asyncio              1.6.0
networkx                  3.3
notebook                  7.2.2
notebook_shim             0.2.4
numpy                     1.26.4
openpyxl                  3.1.5
osmnx                     1.9.4
overrides                 7.7.0
packaging                 24.2
pandas                    2.2.3
pandocfilters             1.5.1
parso                     0.8.4
pillow                    11.0.0
pip                       22.3.1
platformdirs              4.3.6
plumbum                   1.9.0
pooch                     1.8.2
prometheus_client         0.21.0
prompt_toolkit            3.0.48
psutil                    6.1.0
pure_eval                 0.2.3
pyaedt                    0.11.5
pycparser                 2.22
pydantic                  2.9.2
pydantic_core             2.23.4
pyedb                     0.31.0
Pygments                  2.18.0
pyparsing                 3.2.0
pyproj                    3.7.0
python-dateutil           2.9.0.post0
python-json-logger        2.0.7
pytomlpp                  1.0.13
pytz                      2024.2
pyvista                   0.44.1
pywin32                   308
pywinpty                  2.0.14
PyYAML                    6.0.2
pyzmq                     26.2.0
referencing               0.35.1
requests                  2.32.3
rfc3339-validator         0.1.4
rfc3986-validator         0.1.1
rich                      13.9.4
rpds-py                   0.21.0
rpyc                      6.0.1
Rtree                     1.3.0
scikit-learn              1.5.2
scikit-rf                 1.3.0
scipy                     1.14.1
scooby                    0.10.0
Send2Trash                1.8.3
setuptools                65.5.0
shapely                   2.0.6
six                       1.16.0
sniffio                   1.3.1
soupsieve                 2.6
spyder-kernels            3.0.1
SRTM.py                   0.3.7
stack-data                0.6.3
terminado                 0.18.1
threadpoolctl             3.5.0
tinycss2                  1.4.0
toml                      0.10.2
tomli                     2.1.0
tornado                   6.4.1
traitlets                 5.14.3
types-python-dateutil     2.9.0.20241003
typing_extensions         4.12.2
tzdata                    2024.2
uri-template              1.3.0
urllib3                   2.2.3
utm                       0.7.0
vtk                       9.3.1
wcwidth                   0.2.13
webcolors                 24.11.1
webencodings              0.5.1
websocket-client          1.8.0
widgetsnbextension        3.6.10

Samuelopez-ansys, 2024-11-21:
Hi @SK-mechcoder,

Could expand the message manager?

It is working for me:

[image]

[image]

I have the AEDT Patch installed from the Ansys customer portal, but I do not think it is interfering.

SK-mechcoder, 2024-11-22:
This is the message I get in the AEDT message window.
[image]

and this is the message in the terminal.
[image]

Samuelopez-ansys, 2024-11-22:
Hi @Samuelopez-ansys ,

This is not giving any error; it is loading the PyAEDT object, but it is impossible to understand the issue without debugging.

Please check that you have an active license because the license server checks the license when it opens the Maxwell design.

If it still does not work, please contact the Ansys customer portal support, because we will need to debug the issue, don't hesitate to get in touch with the Ansys customer portal support.

Samuelopez-ansys, 2024-11-22:
Hi @SK-mechcoder , 

I think it is a firewall issue. Usually the users disable it or disable the proxy server:

https://aedt.docs.pyansys.com/version/stable/Getting_started/Troubleshooting.html#failure-connecting-to-the-grpc-server

https://github.com/ansys/pyaedt/issues/5141

SK-mechcoder, 2024-11-22:
Hi @Samuelopez-ansys ,

I tried disabling the proxy server. It worked. Thanks for the help.

closed 2024-11-22 by SK-mechcoder