ansys/pyaedt #5479: Add "Single Point" to add_eddy_current_sweep

IreneWoyna-HE, 2024-11-25:
### Description of the feature

Dear pyaedt team!

It seems that I can't add a single point, only single points (corresponding to Single Point Sweep):

[image]

I tried this, but it doesn't add frequency sweep:

sim_setup = m2d.create_setup()
sim_setup.add_eddy_current_sweep(range_type="SinglePoints", start = 0.001)

Thanks in advance for looking into this.

Thanks
Irene

### Steps for implementing the feature

sim_setup = m2d.create_setup()
sim_setup.add_eddy_current_sweep(range_type="SinglePoints", start = 0.001)

gmalinve, 2024-11-29:
@IreneWoyna-HE I'll close this one since I fix the issue in #5492.
It will be available in today's new release. 
Please do let me know!

Giulia

closed 2024-11-29 by gmalinve

IreneWoyna-HE, 2024-11-29:
Many thanks Giulia! I'll test asap.