ansys/pyaedt #5500: Error in Generating and Exporting Field Plots Using PyAEDT

chamanth-vct, 2024-11-29:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

I am encountering issues while generating and exporting field plots using PyAEDT.  The process fails with the following error messages displayed in the AEDT Message Manager as shown in the screenshot:

```
Script macro error: Plot H_vector_plot is not valid. Make sure this plot is updated before export
```
```
case file format is not supported for this plot.
```

[image]

However, when performing the same steps using the AEDT GUI, the process completes successfully without any errors as shown in the GIF below.

[image]

### Steps To Reproduce

Use the provided Python code snippet (below) to reproduce the issue. Download and use the sample aedt file from this [link](https://apps.vcollab.com/fileshare/downloadsharedfile?file_ID=3obabsm42f37cj).

```py

import ansys.aedt.core 
from pathlib import Path
import pyvista as pv

folder_path = Path(r"C:\Users\chamanth\Downloads\New_folder_3")
file_path =  folder_path.joinpath("Project3.aedt")     

exports_folder_path = folder_path.joinpath("exports_folder")
exports_folder_path.mkdir(parents=True,exist_ok=True)

result_folder = exports_folder_path.joinpath("result_folder")
result_folder.mkdir(parents=True,exist_ok=True)

desktop = ansys.aedt.core.Desktop() 
prj= desktop.load_project(file_path.__str__())
prj.set_auto_open(enable=False)
model = prj.modeler
part_names = model.object_names
post = prj.post  # post_processor

part_list = ['Ground'] 

for name in part_list:

    obj =  model.get_object_from_name(name)
    if obj.solve_inside:
        h_field_plot = post.create_fieldplot_volume([name],quantity="Vector_H",
                                                    plot_name="H_vector_plot")
    else:
        h_field_plot = post.create_fieldplot_surface(obj.faces,quantity="Vector_H",
                                                    plot_name="H_vector_plot",
                                                    intrinsics=prj.setups[0].default_intrinsics)

    result_file_name = result_folder.joinpath(name+'_h_vector')
    export_status = post.export_field_plot(plot_name = "H_vector_plot",
                            output_dir = exports_folder_path,
                            file_name = result_file_name.__str__(),
                            file_format = 'case')

    if export_status:
        print(f"\tExported H_Vector")
    else:
        print(f"\tNot Exported H_Vector")

    h_field_plot.delete()

prj.close_project()
desktop.close_desktop()

```

The Python code generates the H Vector field plot for the `Ground` object and attempts to export it into .case format.

Based on this error message: `Script macro error: Plot H_vector_plot is not valid. Make sure this plot is updated before export`
I attempted to update the plot using the following command: `h_field_plot.update()`
Unfortunately, this approach did not resolve the problem.

Could anyone assist in resolving this issue? Any suggestions or insights would be highly appreciated.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

```txt
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
asttokens==2.4.1
attrs==24.2.0
certifi==2024.8.30
cffi==1.17.1
charset-normalizer==3.3.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.2
contourpy==1.3.0
cycler==0.12.1
debugpy==1.8.5
decorator==5.1.1
defusedxml==0.7.1
exceptiongroup==1.2.2
executing==2.1.0
fonttools==4.53.1
fpdf2==2.7.9
idna==3.10
ipykernel==6.29.5
ipython==8.27.0
jedi==0.19.1
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
jupyter_client==8.6.3
jupyter_core==5.7.2
kiwisolver==1.4.7
llvmlite==0.43.0
matplotlib==3.9.2
matplotlib-inline==0.1.7
nest-asyncio==1.6.0
numba==0.60.0
numpy==1.26.4
packaging==24.1
pandas==2.2.2
parso==0.8.4
pillow==10.4.0
platformdirs==4.3.6
plumbum==1.8.3
pooch==1.8.2
prompt_toolkit==3.0.47
psutil==6.0.0
pure_eval==0.2.3
pyaedt==0.10.2
pycparser==2.22
pydantic==2.9.2
pydantic_core==2.23.4
pyedb==0.28.0
Pygments==2.18.0
pyparsing==3.1.4
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.2
pyvista==0.44.1
pywin32==306
PyYAML==6.0.2
pyzmq==26.2.0
referencing==0.35.1
requests==2.32.3
rpds-py==0.20.0
rpyc==6.0.0
Rtree==1.3.0
scikit-rf==1.3.0
scipy==1.14.1
scooby==0.10.0
six==1.16.0
stack-data==0.6.3
toml==0.10.2
tornado==6.4.1
traitlets==5.14.3
typing_extensions==4.12.2
tzdata==2024.1
urllib3==2.2.3
vtk==9.3.1
wcwidth==0.2.13
```

chamanth-vct, 2025-01-09:
Hi  @gmalinve

I hope you’re doing well! 
Just following up on this issue as we’re still facing the problem. Whenever you get a chance, your input would be greatly appreciated.

Thank you!

chamanth-vct, 2025-01-11:
Hi @gmalinve,

Thank you so much for your suggestion—it was really helpful! I’m now able to export it properly.

However, I wanted to point out that the `hfss.setups[0].default_intrinsics` attribute seems to behave inconsistently. For some files, it provides both the frequency and phase, while for others, it only gives the phase.

I haven’t yet identified the difference, but I’ll investigate further and update you if I find anything. If this attribute’s behavior could be made consistent, it would be ideal.

Thanks again for your support!

gmalinve, 2025-01-13:
@chamanth-vct I'll proceed closing this issue since the matter has been solved.
For any other doubt or feedback please feel free to open a new issue.

Thanks!

Giulia

closed 2025-01-13 by gmalinve

gmalinve, 2025-01-13:
@chamanth-vct please look at this issue that I just open: #5650 .

chamanth-vct, 2025-01-24:
Hi @gmalinve,

Based on my testing, when a model contains any encrypted components, the `hfss.setups[0].props` attribute becomes empty.

The method you suggested to retrieve the frequency using the `hfss.setups[0].properties` attribute works well for models with a single frequency solve type. However, for models with a multiple frequency solve type, the solution frequency is represented as "Multiple". As a result, it is not possible to obtain specific frequencies in such cases.

I have raised a new issue to provide more details on this behaviour. You can find it here: https://github.com/ansys/pyaedt/issues/5691