ansys/pyaedt #5506: Add "number of segments" to create_polyline

IreneWoyna-HE, 2024-11-29:
### Description of the feature

Dear pyaedt team,

I'm creating windings using polyline arcs, but I'm missing the possibility to control the number of segments of the polyline arc. Could you please add that? Alternatively / additionally I would happy to know how I could change the number of segments after creating the polyline.

The thing is that I'm creating foil windings, i.e. very thin conductor layers very close to each other, and if I don't have a way to control the no. of segments, solvers throws "mesh leak to the air" or similar error even when I control the mesh appropriately.

Many thanks in advance!

Cheers
Irene

### Steps for implementing the feature

oEditor.CreatePolyline(
	[
		"NAME:PolylineParameters",
		"IsPolylineCovered:="	, True,
		"IsPolylineClosed:="	, False,
		[
			"NAME:PolylinePoints",
			[
				"NAME:PLPoint",
				"X:="			, "160mm",
				"Y:="			, "-300mm",
				"Z:="			, "0mm"
			],
			[
				"NAME:PLPoint",
				"X:="			, "243.5580551423mm",
				"Y:="			, "-237.232952549388mm",
				"Z:="			, "0mm"
			],
			[
				"NAME:PLPoint",
				"X:="			, "304.105244939971mm",
				"Y:="			, "-152.052622469986mm",
				"Z:="			, "0mm"
			]
		],
		[
			"NAME:PolylineSegments",
			[
				"NAME:PLSegment",
				"SegmentType:="		, "AngularArc",
				"StartIndex:="		, 0,
				"NoOfPoints:="		, 3,
				**"NoOfSegments:="	, "0",**
				"ArcAngle:="		, "35.3624618870691deg",
				"ArcCenterX:="		, "0mm",
				"ArcCenterY:="		, "0mm",
				"ArcCenterZ:="		, "0mm",
				"ArcPlane:="		, "XY"
			]
		],
		[
			"NAME:PolylineXSection",
			"XSectionType:="	, "None",
			"XSectionOrient:="	, "Auto",
			"XSectionWidth:="	, "0mm",
			"XSectionTopWidth:="	, "0mm",
			"XSectionHeight:="	, "0mm",
			"XSectionNumSegments:="	, "0",
			"XSectionBendType:="	, "Corner"
		]
	], 
	[
		"NAME:Attributes",
		"Name:="		, "Polyline1",
		"Flags:="		, "",
		"Color:="		, "(143 175 143)",
		"Transparency:="	, 0,
		"PartCoordinateSystem:=", "Global",
		"UDMId:="		, "",
		"MaterialValue:="	, "\"vacuum\"",
		"SurfaceMaterialValue:=", "\"\"",
		"SolveInside:="		, True,
		"ShellElement:="	, False,
		"ShellElementThickness:=", "0mm",
		"ReferenceTemperature:=", "20cel",
		"IsMaterialEditable:="	, True,
		"IsSurfaceMaterialEditable:=", True,
		"UseMaterialAppearance:=", False,
		"IsLightweight:="	, False
	])

gmalinve, 2024-12-03:
Hi @IreneWoyna-HE,

If I understood correctly you'd like to draw your geometry by providing different polylines.
If this is your intention you can achieve it through the _create_polyline_ method because in the _segment_type_ argument you can provide a list of PolylineSegment objects.
In the following example I provide different polylines of different types by giving a list of points as a reference:
```
ref_points = [
    [0.4, 0, 0],
    [-0.4, -0.6, 0],
    [0.4, 0, 0]]

my_geometry = m2d.modeler.create_polyline(
    points=ref_points,
    segment_type=[PolylineSegment(segment_type="AngularArc", arc_center=[0, 0, 0], arc_angle="180deg", arc_plane="XY"),
                  PolylineSegment(segment_type="Line"),
                  PolylineSegment(segment_type="AngularArc", arc_center=[0, -0.6, 0], arc_angle="180deg",
                                  arc_plane="XY"),
                  PolylineSegment(segment_type="Line")],
)
```
A small comment with this approach in case you have some angular arc, you should provide only the initial point of your arc because the method automatically computes the other two points that define the arc. The second point that you provide is the second point of the next polyline (in my case is the second point of my line).

The other option could be drawing line by line you geometry with the _create_polyline_ method and at the end unite and cover the polylines to get a unique geometry:
```
unite_obj = m2d.modeler.unite(assignment=[poly1, poly2, ... ])
covered_obj = m2d.modeler.cover_lines(unite_obj)
```
The drawback of this way is that, in case of many geometries, is very time consuming.

Hope this helps,
In case I misunderstood your intention let me know!

Giulia

IreneWoyna-HE, 2024-12-17:
Thanks @gmalinve! This solves my problem:

segment_type=PolylineSegment(segment_type="Arc",
                                                                arc_center=[0, 0, 0],
                                                                arc_angle="180deg",
                                                                arc_plane="XY",
                                                                num_seg=100)

closed 2024-12-17 by IreneWoyna-HE