ansys/pyaedt #5530: Add support for more than one circuit schematic page

DaveTwyman, 2024-12-04:
### Description of the feature

Presently circuit schematic (Maxwell circuit, circuit, Simplorer etc) only allow one schematic page. Add support for having multiple pages. 
Needed for ETK, Electronic Transformer Toolkit.

### Steps for implementing the feature

Expose page number create_component method of primitives_circuit.py to PyAEDT

gmalinve, 2024-12-05:
@DaveTwyman is this not what you have already implemented with the create_page in this PR #5283?

DaveTwyman, 2024-12-05:
@gmalinve , Not quite, when circuit elements are added the page number needs setting to have them land on the intended page if there is more than one page. UI and native API behave differently in this respect.

At the moment Pyaedt just supports all circuit elements landing on first page

DaveTwyman, 2024-12-12:
@gmalinve , Thinking about this a bit more, it would make sense if this is added at the same time as the other circuit components in R https://github.com/ansys/pyaedt/pull/5283 as you've pointed out.

I'll close this PR and create the new class in Schematic as discussed earlier on the Office Hours call.

closed 2024-12-12 by DaveTwyman