ansys/pyaedt #5567: `import_gds_3d` doesn't work in non graphical mode

geexie, 2024-12-13:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

I started using PyAEDT to avoid running over GUI. I discovered that `import_gds_3d` doesn't work in non graphical mode. Is it possible to overcome this issue?

### Steps To Reproduce

m3d = Maxwell3d(project=project_name, version="2024.1", new_desktop=True, non_graphical=True, close_on_exit=True)
mapping_layers = {1: (0, 100)}
if m3d.import_gds_3d(os.path.join(proj_dir, "sim.GDS"), mapping_layers, units='um', import_method=0):
    print("export is successful")

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.11

### Installed packages

python -m pip freeze
aiohappyeyeballs==2.4.3
aiohttp==3.11.8
aiosignal==1.3.1
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc4
ansys-tools-visualization-interface==0.6.0
async-timeout==5.0.1
attrs==24.2.0
blosc2==2.7.1
certifi==2024.8.30
cffi==1.17.1
charset-normalizer==3.4.0
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
contourpy==1.3.1
cycler==0.12.1
defusedxml==0.7.1
distro==1.9.0
dotnetcore2==3.1.23
et_xmlfile==2.0.0
fast_simplification==0.1.7
fiona==1.10.1
fonttools==4.55.0
fpdf2==2.8.1
frozenlist==1.5.0
geopandas==0.14.4
idna==3.10
imageio==2.36.1
joblib==1.4.2
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
kiwisolver==1.4.7
markdown-it-py==3.0.0
matplotlib==3.9.2
mdurl==0.1.2
meshio==5.3.5
more-itertools==10.5.0
msgpack==1.1.0
multidict==6.1.0
ndindex==1.9.2
networkx==3.3
numexpr==2.10.2
numpy==1.26.4
openpyxl==3.1.5
osmnx==1.9.4
packaging==24.2
pandas==2.2.3
pillow==11.0.0
platformdirs==4.3.6
plumbum==1.9.0
pooch==1.8.2
propcache==0.2.0
psutil==6.1.0
py-cpuinfo==9.0.0
pyaedt==0.11.6
pycparser==2.22
pydantic==2.9.2
pydantic_core==2.23.4
pyedb==0.31.0
Pygments==2.18.0
pyparsing==3.2.0
pyproj==3.7.0
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.2
pyvista==0.44.2
PyYAML==6.0.2
referencing==0.35.1
requests==2.32.3
rich==13.9.4
rpds-py==0.21.0
rpyc==6.0.1
Rtree==1.3.0
scikit-learn==1.5.2
scikit-rf==1.4.1
scipy==1.14.1
scooby==0.10.0
shapely==2.0.6
six==1.16.0
SRTM.py==0.3.7
tables==3.10.1
threadpoolctl==3.5.0
toml==0.10.2
trame==3.7.0
trame-client==3.5.0
trame-server==3.2.3
trame-vtk==2.8.12
trame-vuetify==2.7.2
typing_extensions==4.12.2
tzdata==2024.2
urllib3==2.2.3
utm==0.7.0
vtk==9.3.1
websockets==14.1
wslink==2.2.1
yarl==1.18.0

geexie, 2024-12-16:
@Samuelopez-ansys, thank you! Looking forward to see if there is a solution

closed 2024-12-16 by Samuelopez-ansys

geexie, 2024-12-18:
I tried this fix with ansys 2024.2 and for my example this version crushes. Could you please check if this works on you side?

I use the following GDS [file](https://github.com/geexie/moonmon/blob/main/moonmon_460_for_sim.GDS) 

My code to reproduce the issue

```
import os
import shutil
import time
from ansys.aedt.core import Maxwell3d

rerun = True

# Create an instance of the ``Maxwell3d``.
proj_dir = os.path.join(os.getcwd(), "data3")

if rerun:
    shutil.rmtree(proj_dir)
    if not os.path.exists(proj_dir):
        os.makedirs(proj_dir)
    print ("proj_dir = ", proj_dir)

project_name = os.path.join(proj_dir, "capacitance.aedt")
m3d = Maxwell3d(project=project_name, version="2024.2", new_desktop=True, non_graphical=True, close_on_exit=True)

m3d.solution_type = m3d.SOLUTIONS.Maxwell3d.ElectroStatic

mapping_layers = {1: (0, 100)}
if m3d.import_gds_3d(os.path.join(proj_dir, "../../moonmon/moonmon_460_for_sim.GDS"),
                     mapping_layers, units='nm', import_method=0):
    print("export is successful")

# check if objects are exported
objects_names = m3d.modeler.model_objects
print("objects = ", objects_names)

objects = m3d.modeler.object_list
print("objects = ", objects)

# Save everything
m3d.save_project()
```

I got an empty object list with this code both running natively and remotely over SSH

```
(pyaedt-dev) geexie@Tower:~/code/chip/QmQ$ python3 m3d.py 
proj_dir =  /home/geexie/code/chip/QmQ/data3
PyAEDT INFO: Python version 3.10.12 (main, Nov  6 2024, 20:22:13) [GCC 11.4.0]
PyAEDT INFO: PyAEDT version 0.13.dev0.
PyAEDT INFO: Initializing new Desktop session.
PyAEDT INFO: Log on console is enabled.
PyAEDT INFO: Log on file /tmp/pyaedt_geexie_16b666b1-339e-45bf-8de7-33266527f91a.log is enabled.
PyAEDT INFO: Log on AEDT is enabled.
PyAEDT INFO: Debug logger is disabled. PyAEDT methods will not be logged.
PyAEDT INFO: Launching PyAEDT with gRPC plugin.
PyAEDT INFO: New AEDT session is starting on gRPC port 50871
PyAEDT INFO: AEDT installation Path /opt/AnsysEM/v242/Linux64/
PyAEDT INFO: Ansoft.ElectronicsDesktop.2024.2 version started with process ID 2292068.
PyAEDT INFO: Project capacitance has been created.
PyAEDT INFO: No design is present. Inserting a new design.
PyAEDT INFO: Added design 'Maxwell 3D_6AY' of type Maxwell 3D.
PyAEDT INFO: Aedt Objects correctly read
PyAEDT INFO: GDS layer imported with elevations and thickness.
export is successful
PyAEDT INFO: Modeler class has been initialized! Elapsed time: 0m 0sec
objects =  []
objects =  []
PyAEDT INFO: Project None Saved correctly
```

Ansys shows crush dialogue window
[image]

geexie, 2024-12-18:
@Samuelopez-ansys, could you please take a look?

Samuelopez-ansys, 2024-12-20:
Hi @geexie ,

In fact, it is a bug in Maxwell 3D, because in HFSS I tried and it worked, but not in Maxwell.

Please, submit a service request to the Ansys support if you want this feature to be implemented in the future.

In the meanwhile, you could use the following workaround, import in HFSS, and copy the solids to Maxwell 3D:

```
import os
import shutil
import time
from ansys.aedt.core import Maxwell3d, Hfss

rerun = True

# Create an instance of the ``Maxwell3d``.
proj_dir = r"C:\Users\slopez\Downloads"

project_name = os.path.join(proj_dir, "capacitance.aedt")
hf = Hfss(project=project_name,
          version="2024.2",
          new_desktop=True,
          non_graphical=False,
          close_on_exit=True,
          )

mapping_layers = {1: (0, 100)}
if hf.import_gds_3d(os.path.join(proj_dir, "moonmon_460_for_sim.GDS"),
                    mapping_layers, units='nm', import_method=0):
    print("export is successful")

m3d = Maxwell3d(solution_type="ElectroStatic")

m3d.copy_solid_bodies_from(hf)

# check if objects are exported
objects_names = m3d.modeler.model_objects
print("objects = ", objects_names)

objects = m3d.modeler.object_list
print("objects = ", objects)

# Save everything
m3d.save_project()

m3d.release_desktop()
```

Finally, if you are using Layout geometries, I would use PyEDB to first manipulate the layout:

https://github.com/ansys/pyedb

geexie, 2024-12-20:
@Samuelopez-ansys, thanks for your help! 

Will use you workaround in a meanwhile. Is it correct that you instantiate HFSS with `non_graphical=False` parameter? Can I use `non_graphical=True,`?  Is it just a typo?

```
hf = Hfss(project=project_name,
          version="2024.2",
          new_desktop=True,
          non_graphical=False,
          close_on_exit=True,
          )
```

geexie, 2024-12-22:
@Samuelopez-ansys, I checked and the workaround works for this short example. However, HFSS import works a bit different. I'm getting an error while I try to assign excitation 

```
PyAEDT ERROR: **************************************************************
PyAEDT ERROR:   File "/home/geexie/code/chip/QmQ/m3d.py", line 59, in <module>
PyAEDT ERROR:     e1 = m3d.assign_voltage(assignment=objects[0], amplitude=0, name="E1")
PyAEDT ERROR:   File "/home/geexie/code/pyaedt-dev/lib/python3.10/site-packages/ansys/aedt/core/maxwell.py", line 1003, in assign_voltage
PyAEDT ERROR:     if bound.create():
PyAEDT ERROR:   File "/home/geexie/code/pyaedt-dev/lib/python3.10/site-packages/ansys/aedt/core/modules/boundary/common.py", line 447, in create
PyAEDT ERROR:     self._app.oboundary.AssignVoltage(self._get_args())
PyAEDT ERROR: AEDT API Error on create
PyAEDT ERROR: Last Electronics Desktop Message - [error]  script macro error: excitation 'e1': cannot apply voltage excitation to non planar entity. (04:50:04 pm  dec 22, 2024)
```

Same when I try to do it manually in Ansys

[image]

Manually, I managed to create a section Signal->Edit->Surface->Section->XY option, then assign voltage to it. Could you please advice me how can I do it with Python? It should be something with [section](https://aedt.docs.pyansys.com/version/stable/API/_autosummary/ansys.aedt.core.modeler.cad.object_3d.Object3d.section.html) but `sect1 = objects[0].section("XY")` gives the same error

Full code example

```
import os
import shutil
import time
from ansys.aedt.core import Maxwell3d, Hfss

rerun = True

# Create an instance of the ``Maxwell3d``.
proj_dir = os.path.join(os.getcwd(), "data3")
if rerun:
    shutil.rmtree(proj_dir)
    if not os.path.exists(proj_dir):
        os.makedirs(proj_dir)
    print ("proj_dir = ", proj_dir)

project_name = os.path.join(proj_dir, "capacitance.aedt")
hf = Hfss(project=project_name,
          version="2024.2",
          new_desktop=True,
          non_graphical=True,
          close_on_exit=True,
          )

design = os.path.join(proj_dir, "../../moonmon/moonmon_460_for_sim.GDS")
if not os.path.exists(design):
    raise Exception ("design file doesn't exist {}".format(design))

# import with HFSS (workaround to export model GUI-less)
mapping_layers = {1: (0, 100)}
if hf.import_gds_3d(design, mapping_layers, units='nm', import_method=0):
    print("export is successful")

# check if objects are exported
objects_names = hf.modeler.model_objects
print("objects = ", objects_names)

objects = hf.modeler.object_list
print("objects = ", objects)

# copy to
m3d = Maxwell3d(solution_type="ElectroStatic")
m3d.copy_solid_bodies_from(hf, include_sheets=True)

# check if objects are exported
objects_names = m3d.modeler.model_objects
print("objects = ", objects_names)

objects = m3d.modeler.object_list
print("objects = ", objects)

# assign material
if m3d.assign_material(objects,"pec"):
    print("pec is assigned")

subtrace = m3d.modeler.create_box(origin=[-1, -1, -0.5], sizes=[2, 2, 0.5], name="subtrace", material="sapphire")
vacuum = m3d.modeler.create_box(origin=[-1, -1, 0], sizes=[2, 2, 2], name="vacuum", material="vacuum")

sect1 = objects[0].section("XY")

# assign excitations
# e1 = m3d.assign_voltage(assignment=objects[0], amplitude=0, name="E1")
e1 = m3d.assign_voltage(assignment=sect1, amplitude=0, name="E1")
# e2 = m3d.assign_voltage(assignment=objects[1], amplitude=0, name="E2")
# e3 = m3d.assign_voltage(assignment=objects[2], amplitude=0, name="E3")

# Save everything
m3d.save_project()
```

PR #5571 (2024-12-16): FIX: Allow non graphical mode for Import GDSII in 3D
## Description
From 2024R1, AEDT API allows the import of GDSII in non-graphical mode.

## Issue linked
Close #5567 

## Checklist
- [x] I have tested my changes locally.
- [x] I have added necessary documentation or updated existing documentation.
- [x] I have followed the coding style guidelines of this project.
- [x] I have added appropriate tests (unit, integration, system).
- [x] I have reviewed my changes before submitting this pull request.
- [x] I have linked the issue or issues that are solved by the PR if any.
- [x] I have agreed w [... cut]
PR #5668 (2025-01-29): FIX: Various improvements to Object3D
Bug when parsing data for object3d. Wrong group assignment and missing transparency.

## Description
**Please provide a brief description of the changes made in this pull request.**

## Issue linked
Fixed #5561
Fixed #5567

## Checklist
- [ ] I have tested my changes locally.
- [ ] I have added necessary documentation or updated existing documentation.
- [ ] I have followed the coding style guidelines of this project.
- [ ] I have added appropriate tests (unit, integration, system).
- [ ] I have reviewed my changes before submitting this pull request.
- [  [... cut]