ansys/pyaedt #5568: Bug located in aedt/core/modules/mesh.py

Devin-Crawford, 2024-12-14:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

There is an error when a ``MeshOperation`` instance is created. An error message states that the child object cannot be found, but execution proceeds without further errors. The AEDT example ``maxwell_icepak.py`` fails when the 2nd Maxwell seeding operation is applied.

It looks like the error was introduced in 
- "FEAT: Use GetDataModel to improve the way information are cached into the API (#5488)"
on Nov 12 on lines 122-125 of ``mesh.py``

### Steps To Reproduce

Run the example ``maxwell_icepak.py`` and you'll see that the 2nd mesh operation fails.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

aiohappyeyeballs==2.4.4
aiohttp==3.11.10
aiosignal==1.3.1
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
ansys-tools-visualization-interface==0.6.1
anyio==4.4.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.1
async-lru==2.0.4
async-timeout==5.0.1
attrs==24.2.0
babel==2.16.0
beautifulsoup4==4.12.3
bleach==6.1.0
blosc2==2.7.1
certifi==2024.7.4
cffi==1.17.0
charset-normalizer==3.3.2
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.2
contourpy==1.2.1
coverage==7.6.9
cycler==0.12.1
debugpy==1.8.5
decorator==5.1.1
defusedxml==0.7.1
et-xmlfile==1.1.0
exceptiongroup==1.2.2
execnet==2.1.1
executing==2.0.1
fast_simplification==0.1.7
fastjsonschema==2.20.0
fiona==1.9.6
fonttools==4.53.1
fpdf2==2.7.9
fqdn==1.5.1
frozenlist==1.5.0
geopandas==0.14.4
h11==0.14.0
httpcore==1.0.5
httpx==0.27.0
idna==3.7
imageio==2.35.1
iniconfig==2.0.0
ipykernel==6.29.5
ipython==8.26.0
ipywidgets==8.1.5
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.4
joblib==1.4.2
json5==0.9.25
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
jupyter==1.0.0
jupyter-console==6.6.3
jupyter-events==0.10.0
jupyter-lsp==2.2.5
jupyter_client==8.6.2
jupyter_core==5.7.2
jupyter_server==2.14.2
jupyter_server_terminals==0.5.3
jupyterlab==4.2.4
jupyterlab_pygments==0.3.0
jupyterlab_server==2.27.3
jupyterlab_widgets==3.0.13
jupytext==1.16.4
kiwisolver==1.4.5
markdown-it-py==3.0.0
MarkupSafe==2.1.5
matplotlib==3.8.4
matplotlib-inline==0.1.7
mdit-py-plugins==0.4.1
mdurl==0.1.2
meshio==5.3.5
mistune==3.0.2
mock==5.1.0
more-itertools==10.5.0
msgpack==1.1.0
multidict==6.1.0
nbclient==0.10.0
nbconvert==7.16.4
nbformat==5.10.4
ndindex==1.9.2
nest-asyncio==1.6.0
networkx==3.3
notebook==7.2.1
notebook_shim==0.2.4
numexpr==2.10.2
numpy==1.26.4
openpyxl==3.1.5
osmnx==1.9.4
overrides==7.7.0
packaging==24.1
pandas==2.2.2
pandocfilters==1.5.1
parso==0.8.4
pillow==10.4.0
platformdirs==4.2.2
pluggy==1.5.0
plumbum==1.8.3
pooch==1.8.2
prometheus_client==0.20.0
prompt_toolkit==3.0.47
propcache==0.2.1
psutil==6.0.0
pure_eval==0.2.3
py-cpuinfo==9.0.0
-e git+https://github.com/ansys/pyaedt.git@f0c5a831f24b76cba5acb0dd0cbcf946903d4327#egg=pyaedt
pycparser==2.22
pydantic==2.8.2
pydantic_core==2.20.1
pyedb==0.24.0
Pygments==2.18.0
pyparsing==3.1.2
pyproj==3.6.1
pytest==8.3.4
pytest-cov==6.0.0
pytest-xdist==3.6.1
python-dateutil==2.9.0.post0
python-json-logger==2.0.7
pytomlpp==1.0.13
pytz==2024.1
pyvista==0.44.1
pywin32==306
pywinpty==2.0.13
PyYAML==6.0.2
pyzmq==26.2.0
qtconsole==5.5.2
QtPy==2.4.1
referencing==0.35.1
requests==2.32.3
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rich==13.7.1
rpds-py==0.20.0
rpyc==6.0.0
Rtree==1.3.0
scikit-learn==1.5.1
scikit-rf==1.2.0
scipy==1.14.1
scooby==0.10.0
Send2Trash==1.8.3
shapely==2.0.6
six==1.16.0
sniffio==1.3.1
soupsieve==2.6
SRTM.py==0.3.7
stack-data==0.6.3
tables==3.10.1
terminado==0.18.1
threadpoolctl==3.5.0
tinycss2==1.3.0
toml==0.10.2
tomli==2.0.1
tomli_w==1.1.0
tornado==6.4.1
traitlets==5.14.3
trame==3.7.1
trame-client==3.5.0
trame-server==3.2.3
trame-vtk==2.8.12
trame-vuetify==2.7.2
types-python-dateutil==2.9.0.20240821
typing_extensions==4.12.2
tzdata==2024.1
uri-template==1.3.0
urllib3==2.2.2
utm==0.7.0
vtk==9.3.1
wcwidth==0.2.13
webcolors==24.8.0
webencodings==0.5.1
websocket-client==1.8.0
websockets==14.1
widgetsnbextension==4.0.13
wslink==2.2.1
yarl==1.18.3

Devin-Crawford, 2024-12-14:
@maxcapodi78 : I confirm that the error is gone if I revert to f0c5a831

SMoraisAnsys, 2024-12-16:
@Devin-Crawford The PR is not yet merged, you should only close an issue when the fix is merged in main :)

SMoraisAnsys, 2024-12-17:
Also, if you add the following text to your PR description, this issue should be automatically closed one the PR is merged

Close #5568

SMoraisAnsys, 2024-12-19:
Fixed by https://github.com/ansys/pyaedt/pull/5596, thanks for warning us about the example failure

PR #5595 (2024-12-20): FIX: Object oriented name update
## Description
**Please provide a brief description of the changes made in this pull request.**

## Issue linked
Close #5568 

## Checklist
- [x] I have tested my changes locally.
- [x] I have added necessary documentation or updated existing documentation.
- [x] I have followed the coding style guidelines of this project.
- [x] I have added appropriate tests (unit, integration, system).
- [x] I have reviewed my changes before submitting this pull request.
- [x] I have linked the issue or issues that are solved by the PR if any.
- [x] I have agreed with the Con [... cut]
closed 2024-12-19 by SMoraisAnsys