ansys/pyaedt #5588: ipk.post.export_field_jpg() function  causes the AEDT Icepak solution result file to disappear.

aknsj, 2024-12-18:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

Hi, I am using the following function when exporting a surface temperature plot.
```
def create_surface_temp_plot(plot_name,obj_list,orientation):
    if not isinstance(plot_name, str):
        raise TypeError("Plot name must be a string")
    surface_list = []
    for obj in obj_list:
        surface_list.extend(ipk.modeler.get_object_faces(obj))

    ipk.modeler.fit_all()
    ipk.post.create_fieldplot_surface(
        surface_list,
        quantityName="SurfTemperature",
        plot_name=plot_name
    )

    png_name = str(plot_name)+".png"
    ipk.post.export_field_jpg(fileName=os.path.join(project_path,png_name),plotName=plot_name,foldername = None,orientation=orientation,selections=obj_list)
    ipk.post.delete_field_plot(plot_name)
```
I have carried out the following tests to reproduce the problem.
Feel free to contact me if you need additional information.
I look forward to your reply, thank you!

### Steps To Reproduce

As you can see below, when using this function to process the following project with only two solids, the following message appeared in Message Manager and then the result file in the GUI interface disappeared.
[image]
[image]
Note: IcepakDesign1.results folder still retains the results file in the .sd file, it looks like the folder and the GUI seem to have lost the link.
[image]
After testing, it was found that create plot in the function can run successfully.
```
ipk.post.create_fieldplot_surface(
    surface_list,
    quantityName="SurfTemperature",
    plot_name=plot_name
)
```
Then, it was found that this step when exporting plot caused the error that the result file disappeared.
```
ipk.post.export_field_jpg(fileName=os.path.join(project_path,png_name),plotName=plot_name,foldername = None,orientation=orientation,selections=obj_list)
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

Package                               Version
------------------------------------- -------------------
absl-py                               2.1.0              
aiofiles                              23.1.0             
aiohttp                               3.8.4              
aiosignal                             1.3.1              
alabaster                             0.7.12             
altair                                4.2.2              
anaconda-client                       1.7.2              
[... 362 lines cut ...]
vtk                                   9.3.0
watchdog                              1.0.2
watchfiles                            0.22.0
wcwidth                               0.2.5
webencodings                          0.5.1
websockets                            10.4
Werkzeug                              1.0.1
wheel                                 0.36.2
widgetsnbextension                    3.5.1
win-inet-pton                         1.1.0
win-unicode-console                   0.5
wincertstore                          0.2
wrapt                                 1.12.1
wsproto                               1.2.0
xlrd                                  2.0.1
XlsxWriter                            1.3.8
xlwings                               0.23.0
xlwt                                  1.3.0
xmltodict                             0.12.0
yapf                                  0.31.0
yarl                                  1.8.2
zict                                  2.0.0
zipp                                  3.4.1
zope.event                            4.5.0
zope.interface                        5.3.0

maxcapodi78, 2024-12-20:
@aknsj I've tested with latest version of PyAEDT and the solution is not lost. I've checked and it seems you are using a very old version of pyaedt (0.9.10). Maybe you need to update to latest version of PyAEDT. Next monday 0.13.0 will be released and you could probably benefit of it

closed 2024-12-20 by maxcapodi78

Samuelopez-ansys, 2024-12-20:
Also I recall there were some problems in old AEDT releases, please test 2024R2.

aknsj, 2024-12-22:
Thank you for your replies. I really appreciate it as I guess it might be Christmas holiday time now.
I have updated the pyaedt version to the latest version 0.13.0.
As you can see below, the Ansys Electronics Desktop version is 2024R2.1 and the python version is 3.8.8. However, based on the previously tested project, the same error still occurs. So it seems that this issue cannot be solved by upgrading the pyaedt version.
Also, I found that this issue occurs in some projects and not in some. Maybe I need some time to collect test results from different projects in the same environment and sort out the projects before contacting you.

[image]

aknsj, 2024-12-22:
Thank you for your replies. I really appreciate it as I guess it might be Christmas holiday now.
I have updated the pyaedt version to the latest version 0.13.0.
As you can see below, the Ansys Electronics Desktop version is 2024R2.1 and the python version is 3.8.8. However, based on the previously tested project, the same error still occurs. So it seems that this issue cannot be solved by upgrading the pyaedt version.
Also, I found that this issue occurs in some projects and not in some. Maybe I need some time to collect test results from different projects in the same environment and sort out the projects before contacting you.

[image]