ansys/pyaedt #5610: Can not change Maxwell transient analysis time from pyaedt function

ShinsukeOkayasu, 2024-12-27:
The following issues have been identified in the context of Maxwell transient analysis using PyAEDT:

1. **Time Variable Associated with Field Plots**

   - In addition to the time variable used in Maxwell Transient Analysis, field plots contain their own time variable.
   - Currently, this field plot time variable cannot be directly modified through PyAEDT.

2. **Conflict with Script Variables**

   - It is possible to modify the field plot time variable using script variables.
   - However, in such cases, the variables managed internally by PyAEDT are not updated, resulting in inconsistencies between the internal state and the time variable of the field plots.

3. **Time Variables Distributed Across Plots**

   - Each plot has its own individual time variable, which complicates the post-process workflow for transient analysis and makes it difficult to manage efficiently.

#### Requests and Proposals

1. **Enhancement of PyAEDT Functions**

   - We request the addition of functions or options in PyAEDT to directly modify the time variables used for post-processing.
   - This improvement would resolve inconsistencies between internal states and external operations, enabling more efficient analysis and post-processing.

2. **Unified Management Policy for Plots**

   - To address the issue of time variables being distributed across plots, we propose the introduction of a shared time variable that can be modified collectively.

The following is a sample code snippet that demonstrates the current method of modifying plot times using script functions. However, the internal time variable in PyAEDT is not updated:

ーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーー
# Change fields display
m3d.post.ofieldsreporter.SetPlotsViewSolutionContext(["my_plot1"], 
                                                     "Setup1 : Transient", 
                                                     "Time={}s".format(t))
ーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーーー

-Whole sample script-
```
# -*- coding: utf-8 -*-
import os
import csv
from pyaedt import Maxwell3d
from pyaedt import constants

# Specify the path to the AEDT project
aedt_file_path = r"C:\work2\demo\pyaedt_training\test_Project1mm_24R2.aedt"
print("aedt_file_is_successfully_read")

# Open the project using PyAEDT's Maxwell3D
m3d = Maxwell3d(projectname=aedt_file_path, specified_version="2024.2", non_graphical=False)

# Do Transient analysis
m3d.analyze(setup='Setup1')

# Set up for post processing
sol = m3d.post.reports_by_category.fields(setup=m3d.nominal_sweep)
data = sol.get_solution_data()
time_steps = data.intrinsics["Time"] # these are each time step of your transient analysis

print("time_step list is {}".format(time_steps))
print("read time steps")

# For each time step create a field plot (in this case a field plot surface)
# And export it as an image (here is exported as a jpg but you can export it as an aedtplt file too).
try:
         
    # Make field plot 
    p1 = m3d.post.create_fieldplot_volume(
        assignment=["Box11"],
        quantity="J_Vector",
        setup='Transient',
        plot_name="my_plot1",
        intrinsics= {'Time':'0.0s'},
        field_type='Fields'
    )

    for time_step in time_steps:
        # Set time 
        t = constants.unit_converter(
            time_step,
            unit_system="Time",
            input_units=data.units_sweeps["Time"],
            output_units="s",
            )
        
        print("Success to convert unit")

        

        # change fields display
        m3d.post.ofieldsreporter.SetPlotsViewSolutionContext(["my_plot1"], 
                                                             "Setup1 : Transient", 
                                                             "Time={}s".format(t))
              

      
 
        print("Time is {}".format(t))
        print("Create Fields Plot!")

        # Construct the file name with the time step (ns) in it
        file_name = "plot_{}ns_test.png".format(t)
        print("file name is {} set".format(t))

        # Export image
        p1.export_image(
            full_path=file_name,
            selections=["Box11"]
        )

        print("{}ns png file is made".format(t))

# Save the AEDT project and release PyAEDT session
except Exception as e:
    # Print error message if something goes wrong
    print(f"An error occurred: {e}")

finally:
    # Always save the project and release the desktop session even if an error occurs
    m3d.save_project()
    m3d.release_desktop()

```

gmalinve, 2025-02-14:
Hi @ShinsukeOkayasu,

You don't have to change the time step after creating the plot but what you can do is to create the field plot within the for loop and export it. This example does exactly what you need: 
https://examples.aedt.docs.pyansys.com/version/dev/examples/low_frequency/general/field_export.html .

You have to pass each time step in intrinsics as:
```
for time_step in time_steps:
    t = ansys.aedt.core.generic.constants.unit_converter(
        time_step,
        unit_system="Time",
        input_units=data.units_sweeps["Time"],
        output_units="ms",
    )
    m3d.post.create_fieldplot_surface(
        assignment=m3d.modeler.objects_by_name["Coil_A2"],
        quantity=quantity[0],
        plot_name="J_{}_ms".format(t),
        intrinsics={"Time": "{}ms".format(t)},
    )
    mean_j_field_export = m3d.post.export_field_plot(
        plot_name="J_{}_ms".format(t),
        output_dir=temp_folder.name,
        file_format="aedtplt",
    )
```

Please let me know if this solves your issue so I can close it.

Thanks,

Giulia

ShinsukeOkayasu, 2025-02-25:
Hi @gmalinve 
Thanks for your reply.
I update the code but "export_field_plot" can't be export the picture.
I would like to extract the figure data. What should I do?
I assume the code is well working except for exporting picture.

```
# -*- coding: utf-8 -*-
import os
import csv
from pyaedt import Maxwell3d
from pyaedt import constants

# Specify the path to the AEDT project
aedt_file_path = r"C:\work2\demo\pyaedt_training\test_Project1mm_24R2.aedt"
print("aedt_file_is_successfully_read")

# Open the project using PyAEDT's Maxwell3D
m3d = Maxwell3d(project=aedt_file_path, version="2024.2", non_graphical=False)

# Do Transient analysis
m3d.analyze(setup='Setup1')

# Set up for post processing
sol = m3d.post.reports_by_category.fields(setup=m3d.nominal_sweep)
data = sol.get_solution_data()
time_steps = data.intrinsics["Time"] # these are each time step of your transient analysis

print("time_step list is {}".format(time_steps))
print("read time steps")

# For each time step create a field plot (in this case a field plot surface)
# And export it as an image (here is exported as a jpg but you can export it as an aedtplt file too).
try:
         
    # Make field plot 
    p1 = m3d.post.create_fieldplot_volume(
        assignment=["Box11"],
        quantity="J_Vector",
        setup='Transient',
        plot_name="my_plot1",
        intrinsics= {'Time':'0.0s'},
        field_type='Fields'
    )

    for time_step in time_steps:
        # Set time 
        t = constants.unit_converter(
            time_step,
            unit_system="Time",
            input_units=data.units_sweeps["Time"],
            output_units="s",
            )
        
        print("Success to convert unit")

        ### new from here
        m3d.post.create_fieldplot_surface(
            assignment=m3d.modeler.objects_by_name["Box11"],
            quantity="J_Vector",
            plot_name="J_{}_ns".format(t),
            intrinsics={"Time":"{}ns".format(t)}
        )

        mean_j_field_export = m3d.post.export_field_plot(
        plot_name="J_{}_ns".format(t),
        output_dir="C:\work2\demo\pyaedt_training",
        file_format="fldplt",
        )

        print("Time is {}".format(t))
        print("Create Fields Plot!")
        ### new until here
        
        """
        # change fields display
        m3d.post.ofieldsreporter.SetPlotsViewSolutionContext(["my_plot1"], 
                                                             "Setup1 : Transient", 
                                                             "Time={}s".format(t))
              
        # Construct the file name with the time step (ns) in it
        file_name = "plot_{}ns_test.png".format(t)
        print("file name is {} set".format(t))

        # Export image
        p1.export_image(
            full_path=file_name,
            selections=["Box11"]
        )

        print("{}ns png file is made".format(t))
        """

# Save the AEDT project and release PyAEDT session
except Exception as e:
    # Print error message if something goes wrong
    print(f"An error occurred: {e}")

finally:
    # Always save the project and release the desktop session even if an error occurs
    m3d.save_project()
    m3d.release_desktop()

```