ansys/pyaedt #5620: Bug located in get_antenna_data

Greesh03-Hon, 2024-12-30:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

Hi,

I am trying to solve optimization and retrieve the analysis solution using the get_antenna_data / get_antenna_ffd_solution_data. Whatever I do, I am not able to get the far field data. I get messages saying:
PyAEDT INFO: Far field sphere 3D is assigned
PyAEDT INFO: Exporting antenna metadata...
PyAEDT INFO: PostProcessor class has been initialized! Elapsed time: 0m 1sec
PyAEDT INFO: Post class has been initialized! Elapsed time: 0m 1sec
PyAEDT INFO: Antenna metadata exported.
PyAEDT INFO: Exporting geometry...
PyAEDT INFO: Exporting embedded element patterns.... Done: 14.825634717941284 seconds

and that is followed by error:
 line 333, in export_farfield
PyAEDT ERROR:     self.__farfield_data = FfdSolutionData(pyaedt_metadata_file)
PyAEDT ERROR:                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PyAEDT ERROR: Wrong farfield file load. on export_farfield
PyAEDT ERROR: Last Electronics Desktop Message - [error] fail to write field file.

**[image]
**

### Steps To Reproduce

The antenna designed is of four element with eight ports. Where the phase optimization is analyzed with mentioning the calculation range of phi, theta and frequency. The setup is defined using advanced and the radiation field is defined for 3-dimensional case. Once the optimization is calculated, I'm trying to get the far field data using get_antenna_data() function.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.12

### Installed packages

aiohappyeyeballs==2.4.3
aiohttp==3.11.8
aiosignal==1.3.1
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc4
ansys-tools-visualization-interface==0.6.0
anyio==4.4.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.1
async-lru==2.0.4
attrs==23.2.0
Babel==2.15.0
bcrypt==4.0.1
beautifulsoup4==4.12.3
bleach==6.1.0
blosc2==2.7.1
certifi==2021.10.8
cffi==1.16.0
chardet==4.0.0
charset-normalizer==2.0.12
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.2
contourpy==1.3.1
cryptography==41.0.5
cycler==0.12.1
debugpy==1.8.2
decorator==5.1.1
defusedxml==0.7.1
et_xmlfile==2.0.0
executing==2.0.1
fast_simplification==0.1.7
fastjsonschema==2.20.0
fiona==1.10.1
fonttools==4.55.0
fpdf==1.7.2
fpdf2==2.8.2
fqdn==1.5.1
frozenlist==1.5.0
ftd2xx==1.1.2
future==0.18.2
geopandas==0.14.4
gitdb==4.0.9
GitPython==3.1.27
h11==0.14.0
httpcore==1.0.5
httpx==0.27.0
idna==3.3
imageio==2.36.1
intelhex==2.3.0
ipykernel==6.29.5
ipython==8.26.0
ipywidgets==8.1.5
iso8601==2.1.0
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.4
joblib==1.4.2
json5==0.9.25
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
jupyter-events==0.10.0
jupyter-lsp==2.2.5
jupyter_client==8.6.2
jupyter_core==5.7.2
jupyter_server==2.14.1
jupyter_server_terminals==0.5.3
jupyterlab==4.2.3
jupyterlab_pygments==0.3.0
jupyterlab_server==2.27.2
jupyterlab_widgets==3.0.13
kiwisolver==1.4.7
lxml==4.9.3
markdown-it-py==3.0.0
MarkupSafe==2.1.5
matlab==0.1
matplotlib==3.9.2
matplotlib-inline==0.1.7
mdurl==0.1.2
meshio==5.3.5
mistune==3.0.2
more-itertools==10.5.0
msgpack==1.1.0
multidict==6.1.0
nbclient==0.10.0
nbconvert==7.16.4
nbformat==5.10.4
ndindex==1.9.2
nest-asyncio==1.6.0
networkx==3.3
notebook==7.2.1
notebook_shim==0.2.4
numexpr==2.10.2
numpy==1.26.2
openpyxl==3.1.5
osmnx==1.9.4
overrides==7.7.0
packaging==24.1
pandas==2.2.3
pandocfilters==1.5.1
paramiko==3.3.1
parso==0.8.4
Pillow==10.1.0
platformdirs==4.2.2
plumbum==1.9.0
pooch==1.8.2
prometheus_client==0.20.0
prompt_toolkit==3.0.47
propcache==0.2.0
psutil==6.0.0
pure-eval==0.2.2
py-cpuinfo==9.0.0
pyadi-iio==0.0.16
pyaedt==0.11.6
pybrm_if @ file:///C:/Users/H569048/Downloads/90600936-REV-B/pybrm_if-1.3.3.tar.gz#sha256=56d4e393736d287bb2b27b28d2f9a452e2d77b000c1cd7715d3acccf3d3a2e93
pycparser==2.21
pydantic==2.9.2
pydantic_core==2.23.4
pyedb==0.31.0
Pygments==2.18.0
pykml==0.2.0
pylibiio==0.25
PyNaCl==1.5.0
pyparsing==3.2.0
pyproj==3.7.0
PyQRCode==1.2.1
pyserial==3.5
python-dateutil==2.9.0.post0
python-json-logger==2.0.7
pytz==2024.2
pyvista==0.44.2
pywin32==306
pywinpty==2.0.13
PyYAML==6.0.1
pyzmq==26.0.3
referencing==0.35.1
requests==2.32.3
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rich==13.9.4
rpds-py==0.19.0
rpyc==6.0.1
Rtree==1.3.0
scikit-learn==1.5.2
scikit-rf==1.4.1
scipy==1.14.1
scooby==0.10.0
Send2Trash==1.8.3
setuptools==70.3.0
shapely==2.0.6
six==1.16.0
smmap==5.0.0
sniffio==1.3.1
soupsieve==2.5
spur==0.3.23
SRTM.py==0.3.7
stack-data==0.6.3
tables==3.10.1
terminado==0.18.1
threadpoolctl==3.5.0
tinycss2==1.3.0
toml==0.10.2
tornado==6.4.1
traitlets==5.14.3
trame==3.7.0
trame-client==3.5.0
trame-server==3.2.3
trame-vtk==2.8.12
trame-vuetify==2.7.2
types-python-dateutil==2.9.0.20240316
typing_extensions==4.12.2
tzdata==2024.2
uri-template==1.3.0
urllib3==1.26.9
utm==0.7.0
vtk==9.3.1
wcwidth==0.2.13
webcolors==24.6.0
webencodings==0.5.1
websocket-client==1.8.0
websockets==14.1
widgetsnbextension==4.0.13
wslink==2.2.1
yarl==1.18.0

Devin-Crawford, 2025-01-04:
Hi @Greesh03-Hon, 

Please try using the ``Hfss.get_antenna_data()`` method to first retrieve the data. Then you can create an instance of the ``FfdSolutionData`` class using the metadata file name as an argument. The ``FfdSolutionData`` class works with embedded element patterns from disk and is very fast so you can quickly optimize the array excitations for beam forming. Take a look at [this example](https://examples.aedt.docs.pyansys.com/version/dev/examples/high_frequency/antenna/array.html) and let me know if this works for you.

Devin-Crawford, 2025-01-07:
I can reproduce the issue. This only happens for the discrete sweep. Will continue to investigate.
``` python
array_definition = general_methods.read_json(
    os.path.join(path_to_3dcomp, "array_simple_2x2.json")
)
array_definition["Circ_Patch_5GHz1"] = os.path.join(path_to_3dcomp, "Circ_Patch_5GHz.a3dcomp")

array = hfss.add_3d_component_array_from_json(array_definition) # Simple 2x2 array
array.visible = True
setup = hfss.create_setup(name="Setup", setup_type="HFSSDriven", Frequency="5GHz", MaximumPasses=3)
sweep = setup.add_sweep(name="Sweep", sweep_type="Discrete")
sweep.props["RangeStart"] = "4.5GHz"
sweep.props["RangeEnd"] = "5.5GHz"
sweep.props["RangeCount"] = 3
sweep.update()

hfss.analyze(setup=setup.name)

ffdata = hfss.get_antenna_data(frequencies=sweep.frequencies, setup=hfss.existing_analysis_sweeps[1], sphere="Infinite Sphere1")
```
This results in the same error. Calling ``hfss.get_antenna_data()`` without arguments works fine.

Devin-Crawford, 2025-01-07:
The error occurs when values in the frequency list for the discrete sweep are numeric. If they're passed as strings with units or no argument is passed for frequencies, everything seems ok. @Greesh03-Hon can you confirm this behavior?

The following commands work fine:
``` python
ffdata = hfss.get_antenna_data(setup=hfss.existing_analysis_sweeps[1], sphere="Infinite Sphere1")
```
or
``` python
ffdata = hfss.get_antenna_data(frequencies=["4.5GHz", "5.0GHz", "5.5GHz"], setup=hfss.existing_analysis_sweeps[1], sphere="Infinite Sphere1")
```
but this fails:
``` python
ffdata = hfss.get_antenna_data(frequencies=[4.5, 5.0, 5.5], setup=hfss.existing_analysis_sweeps[1], sphere="Infinite Sphere1")
```

Devin-Crawford, 2025-01-07:
@Samuelopez-ansys : What is the expected behavior of SweepHFSS.frequencies? Currently it returns numeric values like [4.5, 5.0, 5.5] although the frequency units are GHz. Should this return string values? ["4.5GHz", "5.0GHz", "5.5GHz"] or numeric values [4.5E9, 5.0E9, 5.5E9]?

Greesh03-Hon, 2025-01-08:
Hello @Devin-Crawford ,

I have tried out the example you shared as well. The only differences I can see between the model I'm working and the example are:
1. The solution type Im using is 'Modal' where as the example took 'terminal' as default.
2. For the model Im working, I have defined PML radiation boundaries.
Do you think any of these are affecting get_antenna_data function?

Devin-Crawford, 2025-01-09:
Hi @Greesh03-Hon,

I doubt that either of those differences are causing the issue.

Since I found one defect in PyAEDT associated with this submission this issue will be closed when the PR is approved. It'll be impossible for me to go any further without seeing your specific model and script. Could you please submit a defect to the Ansys customer service portal and add a note that it should be assigned to me?  Please also refer to this GitHub issue when you submit the issue.

-Devin

Greesh03-Hon, 2025-01-10:
Thanks for the instructions @Devin-Crawford 
I will be on leave and my work is continued by one my colleague (Bednarsky, Vojtech)
I did notify him the details you shared. He will be submitting the defect in ANSYS customer portal.
Thank you once again for your help!

PR #5632 (2025-01-14): FIX: 5620 far field data
## Description
Resolve issue when numeric values are past for frequencies to ``Hfss.get_antenna_data()``

## Issue linked
Close #5620

## Checklist
- [x] I have tested my changes locally.
- [x] I have added necessary documentation or updated existing documentation.
- [x] I have followed the coding style guidelines of this project.
- [x] I have added appropriate tests (unit, integration, system).
- [x] I have reviewed my changes before submitting this pull request.
- [x] I have linked the issue or issues that are solved by the PR if any.
- [x] I have agreed with the Con [... cut]
closed 2025-01-14 by Samuelopez-ansys