ansys/pyaedt #5623: Active Design not set for Hfss.copy_design_from()

Devin-Crawford, 2025-01-03:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

When ``Hfss.copy_design_from(file, "Design")`` is run, the current design is not updated.

### Steps To Reproduce

``` python
wg_tee_name = os.path.join("Examples", "HFSS", "RF Microwave", "Tee.aedt")
wg_tee_name = os.path.join(self.aedtapp.desktop_install_dir, wg_tee_name)
hfss.copy_design_from(wg_tee_name, "TeeModel")
```
If the hfss app was already running with an active design, then ``hfss.design_name`` will not return the new "TeeModel" design.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

All dependencies are installed and up to date.

ToInoJo, 2025-04-01:
This is also present in Q3D.copy_design_from().

Devin-Crawford, 2025-04-02:
The wrong argument was being passed to ``self.set_active_design()`` on line 3644 of design.py.