ansys/pyaedt #5677: add PML method

ricklyon, 2025-01-19:
### Description of the feature

There doesn't appear to be a way to add a perfectly matched layer (PML) to a HFSS design, without resorting to using the GUI. It'd be very helpful to have something similar to the PML Wizard available in the API.

1maslo, 2025-01-19:
I have just used "hfss.create_open_region(frequency="1GHz", boundary="PML", apply_infinite_ground=False)" and it worked just fine for me, see my example below:

from ansys.aedt.core import Hfss

hfss = Hfss(version="2024.2", non_graphical=False, design = "A", project="B", solution_type="Modal")

modeler = hfss.modeler
modeler.create_box([-500, -500, -500], [1000, 1000, 1000], "TestBox")

hfss.create_open_region(frequency="1GHz", boundary="PML", apply_infinite_ground=False)

hfss.save_project(f)
hfss.close_project()

Isn't it the right way?

ricklyon, 2025-01-20:
Yes, that will work, thanks.

closed 2025-01-20 by ricklyon