ansys/pyaedt #5678: Field Calculator GetTopEntryValue(feildcalculator output result)

1maslo, 2025-01-19:
### Description of the feature

Hi, 

I wish to be able to drive the Field Calculator with direct commands and read the top entry at the end. Now I have to create in API new expression, save it, than read it using PyAEDT and than create a plot to read the values from it. Not really straigh forward. Or am I doing something wrong? 

Thanks!
Maslo

Samuelopez-ansys, 2025-02-03:
Hi @1maslo ,

Please try the FieldsCalculator class:

https://aedt.docs.pyansys.com/version/stable/API/visualization/_autosummary/ansys.aedt.core.visualization.post.fields_calculator.FieldsCalculator.html#ansys.aedt.core.visualization.post.fields_calculator.FieldsCalculator

1maslo, 2025-02-05:
Hi @Samuelopez-ansys ,

thanks for your reply. I’ve just tried it and have several observations.

While calculating the H-field at a point using the expression "t" = PointValue(Point(MyPoint), Real(<Hx,Hy,Hz>)) I only get the value for the z-direction, not for all directions. See below:
```
In [6]: hfss.post.fields_calculator.evaluate("t")
Out[6]: 'z -4.1407868384113998e-07'
```

when I wrote it to a file:
```
In [9]: hfss.post.fields_calculator.write("t", "C:\\MyValue.fld", hfss.nominal_adaptive)
Out[9]: True
```

then, the output contains all directions.
```
Vector data "t"
x 4.6530287351886285e-01
y -4.3394178150308318e-03
z -4.1407868384113998e-07
```

If I ran it directly in the HFSS calculator manually, the output would be:: 
```
Vec : <0.465302873518863, -0.00433941781503083, -4.1407868384114E-07>
Vec : t

```

Did I miss something? How can I get the full top entry value? Ideally, I’d like to store it in a variable as a vector.

Another observation: The methods listed in the link you shared are probably not up to date.

[image]

therefore, I got an error during the first try.

```
 AttributeError: 'PostProcessor3D' object has no attribute 'FieldsCalculator'
```

Many thanks!
Maslo

1maslo, 2025-02-05:
Hi @Samuelopez-ansys,

one more remark: I tried to read the H value at a point, and when adding a new expression, I had to set the assignment to 'None'. Is it correct? It doesn't accept points. See below:
```
In [3]: my_expression = {"name": "Point_H_Real","description": "H Field at point","design_type": ["HFSS", "Q3D Extracto
   ...: r"],"fields_type": ["Fields", "CG Fields"],"solution_type": "","primary_sweep": "Freq","assignment": "","assign
   ...: ment_type": ["Point"],"operations": ["NameOfExpression('<Hx,Hy,Hz>')","Operation('Real')","EnterPoint('SpinRed'
   ...: )","Operation('PointValue')"],"report": ["Data Table", "Rectangular Plot"],}

In [4]: expr_name = hfss.post.fields_calculator.add_expression(my_expression, assignment="Point1")
PyAEDT INFO: Parsing C:/exp_v01.aedt.
PyAEDT INFO: File C:/exp_v01.aedt correctly loaded. Elapsed time: 0m 0sec
PyAEDT INFO: aedt file load time 0.3119051456451416
PyAEDT INFO: PostProcessor class has been initialized! Elapsed time: 0m 1sec
PyAEDT INFO: Post class has been initialized! Elapsed time: 0m 1sec
PyAEDT INFO: Modeler class has been initialized! Elapsed time: 0m 0sec
PyAEDT ERROR: Assignment type is not correct.

In [5]: expr_name = hfss.post.fields_calculator.add_expression(my_expression, None)

In [6]: pointvalue=hfss.post.fields_calculator.evaluate("Point_H_Real")

In [7]: print(pointvalue)
z -1.0144547423339652e-04

In [8]: pointvalue
Out[8]: 'z -1.0144547423339652e-04'`
```

Is there a way to keep the assignment in an expression as a variable, so that one doesn't have to create a new one and delete the old one for every new point/object? My approach might be completely wrong...

Thanks!
Maslo