ansys/pyaedt #5691: Issue with `Setup.props` Attribute in HFSS Models Containing Encrypted Components

chamanth-vct, 2025-01-24:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

Hi, I am working with two HFSS models where the **Solve Type** is set to **Multiple Frequencies**. The only difference between the two models is:

- One model is normal (unencrypted).
- The other model contains a few encrypted components, while the rest remains the same.

Both models can be downloaded from the [link](https://apps.vcollab.com/fileshare/downloadsharedfile?file_ID=3ob5lgm6a6ffgd), folder structure is shown below:

```
aedt_issue_test_data_two
      ├── with_encryption
      │           └── Project3.aedt
      └── no_encryption
                    └── Project3.aedt
```

### Steps To Reproduce

I aim to create field plots and export them for individual objects. Below is the code snippet I am using:

```py
import ansys.aedt.core 
from pathlib import Path

#folder_path = Path(r"C:\aedt_issue_test_data_two\no_encryption")
folder_path = Path(r"C:\aedt_issue_test_data_two\with_encryption")
file_path =  folder_path.joinpath("Project3.aedt")

exports_folder_path = folder_path.joinpath("exports_folder")
exports_folder_path.mkdir(parents=True,exist_ok=True)

result_folder = exports_folder_path.joinpath("result_folder")
result_folder.mkdir(parents=True,exist_ok=True)

desktop = ansys.aedt.core.Desktop() 
prj= desktop.load_project(file_path.__str__())
prj.set_auto_open(enable=False)
model = prj.modeler
part_names = model.object_names
post = prj.post  # post_processor

#print(prj.setups[0].props)

if prj.setups[0].props['SolveType'] == 'MultiFrequency':
    available_frequencies = prj.setups[0].props['MultipleAdaptiveFreqsSetup']['AdaptAt']

part_list = ['Ground','Pin','Cylinder6','Dielectric'] 

intrinsics = prj.setups[0].default_intrinsics
intrinsics["Freq"] = available_frequencies[0]['Frequency']

for name in part_list:

    obj =  model.get_object_from_name(name)
    
    if obj.solve_inside:
        h_field_plot = post.create_fieldplot_volume([name],quantity="Vector_H",
                                                    plot_name="H_vector_plot",
                                                    intrinsics=intrinsics)
    else:
        h_field_plot = post.create_fieldplot_surface(obj.faces,quantity="Vector_H",
                                                    plot_name="H_vector_plot",
                                                    intrinsics=intrinsics)

    result_file_name = result_folder.joinpath(name+'_h_vector')
    export_status = post.export_field_plot(plot_name = "H_vector_plot",
                            output_dir = exports_folder_path,
                            file_name = result_file_name.__str__(),
                            file_format = 'case')

    if export_status:
        print(f"\tExported H_Vector")
    else:
        print(f"\tNot Exported H_Vector")

    h_field_plot.delete()
prj.close_project()
desktop.close_desktop()

```

### Issue:

The code works perfectly for the unencrypted model. However, when used with the encrypted model, the code fails at line 23, specifically when accessing the `prj.setups[0].props` attribute of the setup object.

### Observations:

- When an encrypted component is present in the model, the `prj.setups[0].props` attribute returns empty dictionary instead of the expected dictionary containing the setup properties ([documentation_link](https://aedt.docs.pyansys.com/version/stable/API/_autosummary/ansys.aedt.core.modules.solve_setup.Setup.props.html#ansys.aedt.core.modules.solve_setup.Setup.props)).

- This prevents retrieval of available frequencies and subsequently breaks any Python scripts that rely on `prj.setups[0].props` for post-processing.

- This behavior has been consistently observed across multiple HFSS models with encrypted components.

### Questions:

- Am I missing any steps or configurations in the code?
- Are there any alternative methods to retrieve frequency data from encrypted models?

Your guidance on this issue would be greatly appreciated.

Thank you

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

```
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc3
asttokens==2.4.1
attrs==24.2.0
certifi==2024.8.30
cffi==1.17.1
charset-normalizer==3.3.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.2
contourpy==1.3.0
cycler==0.12.1
debugpy==1.8.5
decorator==5.1.1
defusedxml==0.7.1
exceptiongroup==1.2.2
executing==2.1.0
fonttools==4.53.1
fpdf2==2.7.9
idna==3.10
ipykernel==6.29.5
ipython==8.27.0
jedi==0.19.1
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
jupyter_client==8.6.3
jupyter_core==5.7.2
kiwisolver==1.4.7
llvmlite==0.43.0
matplotlib==3.9.2
matplotlib-inline==0.1.7
nest-asyncio==1.6.0
numba==0.60.0
numpy==1.26.4
packaging==24.1
pandas==2.2.2
parso==0.8.4
pillow==10.4.0
platformdirs==4.3.6
plumbum==1.8.3
pooch==1.8.2
prompt_toolkit==3.0.47
psutil==6.0.0
pure_eval==0.2.3
pyaedt==0.13.2
pycparser==2.22
pydantic==2.9.2
pydantic_core==2.23.4
pyedb==0.31.0
Pygments==2.18.0
pyparsing==3.1.4
python-dateutil==2.9.0.post0
pytomlpp==1.0.13
pytz==2024.2
pyvista==0.44.1
pywin32==306
PyYAML==6.0.2
pyzmq==26.2.0
referencing==0.35.1
requests==2.32.3
rpds-py==0.20.0
rpyc==6.0.0
Rtree==1.3.0
scikit-rf==1.3.0
scipy==1.14.1
scooby==0.10.0
six==1.16.0
stack-data==0.6.3
toml==0.10.2
tomli==2.2.1
tomli_w==1.1.0
tornado==6.4.1
traitlets==5.14.3
typing_extensions==4.12.2
tzdata==2024.1
urllib3==2.2.3
vtk==9.3.1
wcwidth==0.2.13
```

chamanth-vct, 2025-01-24:
In the case of an encrypted model:

- The `props` attribute is empty.
- From version 0.13.0, a new attribute called `properties` was introduced.

### Single Frequency Solve Type:
For models with a single frequency solve type, the solution frequency can be retrieved using the properties attribute, as highlighted in this comment ([Link](https://github.com/ansys/pyaedt/issues/5500#issuecomment-2580213022)).

### Multiple Frequency Solve Type:
For models with a multiple frequency solve type, the solution frequency is represented as "Multiple". Unfortunately, in such cases, we are unable to retrieve specific frequency values even using the properties attribute.

chamanth-vct, 2025-03-03:
Hi @gmalinve and @Samuelopez-ansys,

I hope you are doing well.
I wanted to follow up on this issue as it has been open for a while.
If you have time, could you please take a look?

Please let me know if you need any further clarification. I'm happy to provide more details if needed.

Thanks

Samuelopez-ansys, 2025-03-27:
Hi @chamanth-vct ,

Sorry for the delay. As you well noticed, props is empty when you have encrypted components, this is because props is based in the .aedt file, I mean that PyAEDT is parsing the .aedt file to get the project information. This approach was needed years ago because AEDT API was covering less information, but in the last releases, the information that can be retrieved from the API is more complete, so PyAEDT is migrating to "properties" which is directly using AEDT API calls, which is also much better because it is syncronize with the project (it is not needed to reread the file everytime that something changes).

So, long story short, you should use **properties**.

Are you sure that prj.setups[0].properties["Solution Freq"] is not returning the frequencies? This is what I see:

[image]

chamanth-vct, 2025-03-27:
Hi @Samuelopez-ansys 

Thanks for your response.

I tried using `properties` instead of `props` when encrypted components are present, but I get a different result, which does not match what you showed in the image.

[image]

I'm using Pyaedt 0.15.2 (installed via pip). Could you specify which version of Pyaedt you are using?

[image]

Thanks!

Samuelopez-ansys, 2025-03-27:
@chamanth-vct Same PyAEDT version, which AEDT version are you using?

I see that you have less options; for instance deltaS is not available. And in my case I have Solution Freq and Solution Freq Type, but you do not have. I guess you are using an old AEDT release, and this has been added. 

As I said, AEDT API is improving each release, and thanks to PyAEDT, we are able to find this "gaps".

chamanth-vct, 2025-03-28:
Hi @Samuelopez-ansys 

I tested on 2024 R1 and 2024 R2 and got this result.
I'll check with the latest release (2025 R1) and get back to you.

Thanks!

chamanth-vct, 2025-04-07:
Hi @Samuelopez-ansys 

I checked with 2025 R1, and I can confirm that Solution Freq and Solution Freq Type are available in the properties.

Thank you for the update

closed 2025-04-07 by chamanth-vct