ansys/pyaedt #5709: Improve Setup classes

Devin-Crawford, 2025-01-31:
### Description of the current limitations

The ``SetupHFSS``, ``SetupQ3D`` and other setup class definitions have a common method ``add_sweep()`` that has redundant code. This method should be defined in the base class ``CommonSetup`` and be inherited by the other setup classes.

### Code sample expected

The details of implementation will have to consider the nuances of the different syntax in the native API for the different possible sweep setups. See for example the ``add_sweep`` methods in the setups and modifications made in PR #5683 .

### Useful links and references

See ``aedt\core\modules\solve_sweeps.py`` ``SweepMatrix`` or ``SweepHFSS`` class definitions.

SMoraisAnsys, 2025-02-05:
@Devin-Crawford I'm planning to create a mixins file which should contain classes with methods shared by multiple classes. Would it make sense to ping you once I've started to see if this issue can be handled in such a way ?

Devin-Crawford, 2025-02-05:
> [@Devin-Crawford](https://github.com/Devin-Crawford) I'm planning to create a mixins file which should contain classes with methods shared by multiple classes. Would it make sense to ping you once I've started to see if this issue can be handled in such a way ?

@SMoraisAnsys:  Yes please!

SMoraisAnsys, 2025-02-18:
@Devin-Crawford Was that issue handled in #5683 ?

Devin-Crawford, 2025-02-19:
> [@Devin-Crawford](https://github.com/Devin-Crawford) Was that issue handled in [#5683](https://github.com/ansys/pyaedt/pull/5683) ?

This still should be addressed at a lower level for Q3D and Maxwell.