ansys/pyaedt #5744: BUG: Extension manager seems to move files from venv instead of copying

svandenb-dev, 2025-02-07:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

Extension manager seems to delete python script files instead of copying.

### Steps To Reproduce

NA

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.11

### Installed packages

NA

SMoraisAnsys, 2025-02-12:
@Marwane-20 here's one of the tasks I'd like you to tackle. The content is light so I expect you to have a look at the documentation first and ask questions later. Any question should be directed to me.

Marwane-20, 2025-02-12:
@SMoraisAnsys i will start looking at the documentation, and keep you posted. Thanks.

SMoraisAnsys, 2025-02-24:
@Marwane-20  Thanks, for the feedback. Do you plan to try implementing a fix for this issue ?

Samuelopez-ansys, 2025-02-24:
@SMoraisAnsys @Marwane-20 Thank you for your review.

About site-packages, you are totally right, indeed this is an issue because from AEDT 2024R2, the interpreter has some pre-installed libraries that conflict with PyAEDT, so I will remove it. 

I am still not able to reproduce what @svandenb-dev mentioned.

Marwane-20, 2025-02-24:
@SMoraisAnsys should i propose the change in a PR, or leave @Samuelopez-ansys handle it :) ?

Samuelopez-ansys, 2025-02-24:
@Marwane-20 Please review if you think this is enough https://github.com/ansys/pyaedt/pull/5820

Marwane-20, 2025-02-24:
@Samuelopez-ansys Great, thanks. I also while at it wanted to correct an error in the `Docstring` of `add_script_to_menu` (path : `src/ansys/aedt/core/workflows/customize_automation_tab.py`).
> “The script will be moved to Personal Lib,”  

but the code actually uses (it copies not moves) :  

    shutil.copy2(script_file, dest_script_path)

PR #5820 (2025-02-24): FIX: Remove site --system-site-packages
## Description

## Issue linked
Close #5744 

## Checklist
- [ ] I have tested my changes locally.
- [ ] I have added necessary documentation or updated existing documentation.
- [ ] I have followed the coding style guidelines of this project.
- [ ] I have added appropriate tests (unit, integration, system).
- [ ] I have reviewed my changes before submitting this pull request.
- [ ] I have linked the issue or issues that are solved by the PR if any.
- [ ] I have agreed with the Contributor License Agreement ([CLA](https://developer.ansys.com/form/cla-acc [... cut]