ansys/pyaedt #5776: Bug located in SolutionData

Devin-Crawford, 2025-02-15:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

There appear to be multiple errors when trying to plot 3D far-field data using ``SolutionData.plot_3d()``
- When no arguments are passed and the ``curve`` is ``db(RealizedGainTheta)`` the ``formula`` is set to ``mag`` which causes all negative numbers to be positive.
- There seems to be an error related to the array dimension in some cases that needs further investigation.

### Steps To Reproduce

While modifying the simple dipole example, the following code
``` python
report_3d = hfss.post.reports_by_category.far_field("db(RealizedGainTheta)",
                                                      disc_sweep.name,
                                                      sphere_name="3D",
                                                      Freq= [center_freq],)

report_3d.report_type = "3D Polar Plot"
report_3d.create(name="Realized Gain (dB)")

report_3d_data = report_3d.get_solution_data()
new_plot = report_3d_data.plot_3d()
```
produces the following error:
``` bash
PyAEDT ERROR: Cannot reshape array of size 32761 into shape (181,91) on plot_3d
PyAEDT ERROR: Last Electronics Desktop Message - [error] cannot reshape array of size 32761 into shape (181,91) on plot_3d
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

All modules in the ``pyproject.toml`` are installed.

Devin-Crawford, 2025-02-15:
In the ``SolutionData`` class, the method ``plot_3d()`` applies the method ``self.mag(curve)`` by default. This, in turn, returns magnitude, which applies ``mag()`` to dB, which only returns positive values and does not return the intended: ``db(RealizedGainTheta)``.

Devin-Crawford, 2025-02-15:
The problem is that ``visualization.report.field.FarField.get_solution_data()`` returns only the trace from the report that was already created. In this case, the underlying data is ``db(RealizedGainTheta)`` so no further operation is required. The method ``plot_3d()`` however, applies ``mag`` to the underlying data which is incorrect.

SMoraisAnsys, 2025-02-23:
Is that issue closes and can it be closed ?