ansys/pyaedt #5786: Bug located in move_face()

YuCLarryWu, 2025-02-18:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

Wrong "Value" and "Evaluated Value" are obtained while assigning parameters to the `offset` argument of `move_face()`. 

### Steps To Reproduce

Run the following script.
```
from pyaedt import Maxwell3d

m3d = Maxwell3d(version='2024.2')
m3d['go'] = "0.2mm"

box = m3d.modeler.create_box((0,0,0), (5,5,5))

m3d.modeler.move_face(box.bottom_face_z, '-(go/2mm)')
```

Try four cases by assigning the followings to the `offset` argument: 
1. '-go/2'
2. '(-go/2)'
3. '-go/2mm'
4. '-(go/2mm)'

The "Value" and "Evaluated Value" will be obtained as the followings respectively.
1. 
[image]
2. 
[image]
3. 
[image]
4. 
[image]

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.12

### Installed packages

annotated-types==0.7.0
ansys-pythonnet==3.1.0rc6
asgiref==3.8.1
asttokens==3.0.0
attrs==24.2.0
certifi==2025.1.31
cffi==1.17.1
charset-normalizer==3.4.1
clr_loader==0.2.7.post0
colorama==0.4.6
comm==0.2.2
contourpy==1.3.1
cycler==0.12.1
debugpy==1.8.12
decorator==5.1.1
defusedxml==0.7.1
Django==5.1.6
executing==2.2.0
fonttools==4.55.8
fpdf2==2.8.2
idna==3.10
ipykernel==6.29.5
ipython==8.32.0
jedi==0.19.2
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
jupyter_client==8.6.3
jupyter_core==5.7.2
kiwisolver==1.4.8
matplotlib==3.10.0
matplotlib-inline==0.1.7
nest-asyncio==1.6.0
numpy==2.2.2
packaging==24.2
pandas==2.2.3
parso==0.8.4
pillow==11.1.0
platformdirs==4.3.6
plumbum==1.9.0
pooch==1.8.2
prompt_toolkit==3.0.50
psutil==6.1.1
pure_eval==0.2.3
pyaedt==0.13.3
pycparser==2.22
pydantic==2.10.6
pydantic_core==2.27.2
pyedb==0.36.0
Pygments==2.19.1
pyparsing==3.2.1
python-dateutil==2.9.0.post0
pytz==2025.1
pyvista==0.44.2
pywin32==308
PyYAML==6.0.2
pyzmq==26.2.1
referencing==0.35.1
requests==2.32.3
rpds-py==0.22.3
rpyc==6.0.1
Rtree==1.3.0
scikit-rf==1.5.0
scipy==1.15.1
scooby==0.10.0
six==1.17.0
sqlparse==0.5.3
stack-data==0.6.3
toml==0.10.2
tomli_w==1.2.0
tornado==6.4.2
traitlets==5.14.3
typing_extensions==4.12.2
tzdata==2025.1
urllib3==2.3.0
vtk==9.3.1
wcwidth==0.2.13

jvela018, 2025-02-21:
@YuCLarryWu ,

- This is not so much of a bug as is it a misuse of the move_face() method. Note that this behavior also appears in AEDT. It's not a PyAEDT related behavior. However, we can explain it through PyAEDT

- Why do I say it's a misuse of the method. The method takes a float value as offset, not a string. Whether we can make it work with strings or not, is beyond the scope of the bug report. Let us dive right into it. In the geometry modeler you'll find the definition of the move_face() method:

[image]

Note that the assignment is supposed to be a list [] and not a tuple(), and the offset takes a float, not a string. 

The reason why the error is popping up is because the logic of the offset takes the float, converts it into a string and applies the model's defined units

[image]

- Let us go back to your test cases and rewrite them to see how AEDT is seeing them
    1.	'-go/2'    => ‘-go/2’+’mm’ = -go/2mm = (0.2mm)/(2mm) = 0.1 [no units] It won’t let you perform the operation with the units cancelled out
    2.	'(-go/2)' => '(-go/2)' + ‘mm’ = (-go/2)mm = evaluates the expression (-go/2) and then converts it to mm by applying mm (1e-3, but the model units are already in mm so you just get 0.0001mm
    3. '-go/2mm' = '-go/2mm' + ‘mm’ = -go/2mmmm = error because there’s no such unit or design variable
    4. '-(go/2mm)' = '-(go/2mm)' + ‘mm’ = -(go/2mm)mm = it evaluates the expression go/0.2mm and then applies the unit = -(0.2mm/2mm)mm = -(0.1)mm = 0.1mm (correct)

- My suggestion is to use units only in the design variables (e.g., m3d['go']), use floats in move_face() and set the model units to "mm" in the global units: https://aedt.docs.pyansys.com/version/stable/API/_autosummary/ansys.aedt.core.modeler.modeler_3d.Modeler3D.model_units.html#model-units. Since you're working with python, you can always create a variable mm = 1e-3, and multiply that one to create 2*mm. This way you'll see it as 2*mm on the python script but it'll be read as 2e-3 on the AEDT side. 

Please, let me know if this helps and if I could go ahead and considered this issue resolved.

BR

/jvela018

YuCLarryWu, 2025-03-06:
Thank you for the explanation in details. I understand how it works inside the PyAEDT package.

closed 2025-03-06 by YuCLarryWu