ansys/pyaedt #5848: Pyaedt get unclassified object

soliton37, 2025-02-27:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

Hello,
I'm working with pyaedt and Hfss.modeler
When I search for unclassified objects using the 'unclassified' method, the method returns a list of 3D objects. But, when I search for this unclassified 3D object using the 'get_object_from_name' method, the method returns None.
Is this behavior normal?
I obtained this unclassified object by subtracting two objects, the smaller one (blank part) is entirely included in the larger one (tool part). Astonishment, the 'subtract' method return True with a unclassified object, is it normal behavior ?
Thanks for your comments.
Emmanuel

### Steps To Reproduce

solver.modeler.subtract(touching_signals, ground_object.name)
unclassified_objects = solver.modeler.unclassified_objects
unclassified_signal_object = solver.modeler.get_object_from_name(unclassified_objects[0])  #  return None

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.10

### Installed packages

PyAEDT version 0.14.0.

jvela018, 2025-03-06:
Hi @soliton37 

- This issue is not so much of a bug as it is a misuse of the get_object_from_name method. 
- Let us look at the implementation of the method

[image]
As you can see, it returns the object as long as they are in the self.object_name property, which is defined as the name of the objects excluding unclassified, points

[image]

The reason for the implementation is that the unclassified are assigned to their own list, to be retrieved through the unclassified_objects property. These unclassified objects appear when you create an invalid geometry with some boolean or edit or sweep operation that is not successful. Hence, in pyaedt they're stored separately to be able to keep track of them and be able to delete them using the delete method. 

- In your case note that you already have the unclassified object, hence you don't need to go backwards to get the object through the name - which btw you're missing in your implementation as you accessed the member of the list (object), but you needed to provide the name. This wouldn't work anyway due to what was explained earlier. An alternative would be just to use the object you already retrieved. Meaning:

```python
solver.modeler.subtract(touching_signals, ground_object.name)
unclassified_objects = solver.modeler.unclassified_objects
unclassified_signal_object = unclassified_objects[0]
```

Please, let me know if this answers your question and if this issue can be considered resolved.

BR,

jvela018

soliton37, 2025-03-06:
Hello Jonathan,

Thank you for time and clear explanation, I appreciate.
Regards.

closed 2025-03-06 by jvela018