ansys/pyaedt #5863: Bug located in List index out of range on create_two_resistor_network_block

chs513, 2025-03-05:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

It throws List index out of range on create_two_resistor_network_block, when I use the method create_two_resistor_network_block.

### Steps To Reproduce

from ansys.aedt.core import Icepak
ipk = Icepak()
ipk.modeler.create_box(origin=[0,0,0],sizes=[1,2,3], name='box')
ipk.modeler.create_box(origin=[5,5,5],sizes=[2,2,2], name='board')

ipk.create_two_resistor_network_block(object_name='box', pcb='board', power='1W', rjb=2, rjc=4)

ipk.release_desktop(close_desktop=False, close_projects=False)

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

annotated-types==0.7.0
ansys-pythonnet==3.1.0rc6
appdirs==1.4.4
asttokens==3.0.0
attrs==24.2.0
autopep8==2.3.2
certifi==2025.1.31
cffi==1.17.1
chardet==4.0.0
charset-normalizer==3.4.1
click==8.1.8
clr_loader==0.2.7.post0
colorama==0.4.6
contourpy==1.3.1
cycler==0.12.1
decorator==5.1.1
defusedxml==0.7.1
exceptiongroup==1.2.2
executing==2.2.0
fonttools==4.55.8
fpdf2==2.8.2
grpcio==1.70.0
idna==2.10
ipython==8.32.0
jedi==0.19.2
Jinja2==3.0.1
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
kiwisolver==1.4.8
Mako==1.3.9
MarkupSafe==3.0.2
matplotlib==3.10.0
matplotlib-inline==0.1.7
numpy==2.2.2
packaging==24.2
pandas==2.2.3
parso==0.8.4
Pillow==8.4.0
platformdirs==4.3.6
plumbum==1.9.0
pooch==1.8.2
prompt_toolkit==3.0.50
psutil==6.1.1
pure_eval==0.2.3
pyaedt==0.15.0
pycodestyle==2.12.1
pycparser==2.22
pydantic==2.10.6
pydantic_core==2.27.2
pyedb==0.37.0
pygame==2.6.1
Pygments==2.19.1
pygubu==0.36.3
pygubu-designer==0.40.2
pyparsing==3.2.1
PyQt6==6.4.2
pyqt6-plugins==6.4.2.2.3
PyQt6-Qt6==6.4.3
pyqt6-tools==6.4.2.3.3
PyQt6_sip==13.10.0
python-dateutil==2.9.0.post0
python-dotenv==1.0.1
pythonnet==3.0.5
pytz==2025.1
pyvista==0.44.2
pywin32==308
PyYAML==6.0.2
qt6-applications==6.4.3.2.3
qt6-tools==6.4.3.1.3
referencing==0.35.1
requests==2.25.1
rpds-py==0.22.3
rpyc==6.0.1
Rtree==1.3.0
scikit-rf==1.5.0
scipy==1.15.1
scooby==0.10.0
screeninfo==0.8.1
six==1.17.0
stack-data==0.6.3
tkdesigner==1.0.7
toml==0.10.2
tomli==2.2.1
tomli_w==1.2.0
traitlets==5.14.3
typing_extensions==4.12.2
tzdata==2025.1
urllib3==1.26.20
vtk==9.3.1
wcwidth==0.2.13

siva-krishnaswamy, 2025-03-18:
Hi @chs513 , We are getting this error, since for a successful creation of network boundary condition, the board and component needs to touch completely and should have a common face. 

If you modify your script such that the board and block are touching and have a common face, then you won't face this error. For example

from ansys.aedt.core import Icepak
ipk = Icepak()
ipk.modeler.create_box(origin=[1,5,5],sizes=[1,2,3], name='box')
ipk.modeler.create_box(origin=[1,1,1],sizes=[1,10,10], name='board')

ipk.create_two_resistor_network_block(object_name='box', pcb='board', power='1W', rjb=2, rjc=4)

ipk.release_desktop(close_desktop=False, close_projects=False)

I just modified the box and board dimension such that it touch. Then the error disappear. We will consider an enhancement to the code to give a meaningful error message.

chs513, 2025-03-19:
Hi @siva-krishnaswamy , if there is an tiny gap between the box and the board, Icepak will throw the error, too. I think the board only provides relative positions for the box, they don't need to be attached together.

```
from ansys.aedt.core import Icepak
ipk = Icepak()
ipk.modeler.create_box(origin=[2.001,5,5],sizes=[1,2,3], name='box')
ipk.modeler.create_box(origin=[1,1,1],sizes=[1,10,10], name='board')

ipk.create_two_resistor_network_block(object_name='box', pcb='board', power='1W', rjb=2, rjc=4)

ipk.release_desktop(close_desktop=False, close_projects=False)
```

siva-krishnaswamy, 2025-03-21:
Hi @chs513 , this method relies on the board and component touching. To select the appropriate faces. Please look at the other method

https://aedt.docs.pyansys.com/version/stable/API/_autosummary/ansys.aedt.core.icepak.Icepak.create_network_blocks.html

which is more general where you need to specify the orientation of network block. 

in create_two_resistor_network_block, user doesn't need to specify orientation as it is determined based on the contact between board and component.