ansys/pyaedt #5865: Add APIs for Icepak validations

chs513, 2025-03-05:
### Description of the feature

[image]

chs513, 2025-03-06:
@siva-krishnaswamy, How to set the option Entity Check Level?

siva-krishnaswamy, 2025-03-07:
@chs513 , The option Entity Check Level is not available in the current version. We will consider it for future enhancement. 

Regards,
Siva

chs513, 2025-03-10:
@siva-krishnaswamy Many thanks, Siva.

siva-krishnaswamy, 2025-03-18:
@chs513 , Entity check level is already available as check_level with default argument None.

edit_design_settings(gravity_dir=0, ambient_temperature=20, perform_validation=False, check_level='None', default_fluid='air', default_solid='Al-Extruded', default_surface='Steel-oxidised-surface', export_monitor=False, export_sherlock=False, export_directory='/home/runner/work/pyaedt/pyaedt/doc/source', gauge_pressure=0, radiation_temperature=20, ignore_unclassified_objects=False, skip_intersection_checks=False)

https://aedt.docs.pyansys.com/version/stable/API/_autosummary/ansys.aedt.core.icepak.Icepak.edit_design_settings.html#ansys.aedt.core.icepak.Icepak.edit_design_settings

since all the required API's are available. I will close this request