ansys/pyaedt #5867: When editing pyaedt code, add the appropriate syntax completion.

chs513, 2025-03-05:
### Description of the current limitations

For example, midpoint attritube for edge.midpoint, position attribute for vertice.position, center attritube for box.top_face_x.center.

### Code sample expected

  from ansys.aedt.core import Icepak
  ipk = Icepak()
  box = ipk.modeler.create_box([0,0,0],[1,2,3])

  for edge in box.edges:
      print('Edge Midpoint = ', edge.midpoint)

  for vertice in box.top_face_x.vertices:
      print('Vertics =', vertice.position)

  print('Face center = ', box.top_face_x.center)

Samuelopez-ansys, 2025-03-06:
Hi @chs513 , please could you specify what are you missing in your description? Thanks

lorenzovecchietti, 2025-03-28:
I think it might be an issue of the editor you're using. I'm using pycharm and tried the get the `midpoint` property and autocompletion is working properly

[image]

lorenzovecchietti, 2025-03-28:
@chs513 so things like `edge.midpoint` are working properly or are these other examples of autocompletions that are not working?

chs513, 2025-03-28:
@lorenzovecchietti Like edge.midpoint, it works fine on its own. But my editor couldn't complete edge.midpoint properly. It was just hindering my code editing efforts

lorenzovecchietti, 2025-03-28:
I tried in pycharm and it is working fine with me
[image]

Can you try with this IDE?

chs513, 2025-03-28:
@lorenzovecchietti Bad news. My editor Emacs cannot complete the code completion. Neither can the editor VS code. Let it be.