ansys/pyaedt #5870: Bug in circuit.excitation_objects property

FgPy93, 2025-03-06:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

When using  the ansys.aedt.core.Circuit module for analyzing and retrieving information from a Circuit project errors occured during the usage of the circuit.excitation_objects property. 

When trying to access impedance property of a exciation object in case it was not the first excitation object an error occurs saying: "TypeError: 'NoneType' object is not subscriptable". 

Further research shows that the impedance property of an excitation_object trys to access the _props dictionary. Particularly the rz and iz property are not existent in the dictionary there. 

In case you have only one Port e. g. "Port1" in the circuit project no error occurs. In my case "Port2" shows an error like described.

A very simply circuit project with 4 Ports and a LIN analysis is enough to reproduce the error. I was using AEDT2024R2. 

[image]

### Steps To Reproduce

# The following code can be used to reproduce the issue
from ansys.aedt.core import Circuit

# use python console in the the AEDT UI
circuit = Circuit()

# derive exciation objects dictionary
ex_obj_dict = circuit.excitation_objects

# the following is thrown:
# PyAEDT INFO: ModelerCircuit class has been initialized! Elapsed time: 0m 0sec
# PyAEDT ERROR: **************************************************************
# PyAEDT ERROR:   File "\\rt60-06v\ansysem\v242\Win64\commonfiles\CPython\3_10\winx64\Release\python\lib\runpy.py", line 196, in _run_module_as_main
# PyAEDT ERROR:     return _run_code(code, main_globals, None,
# PyAEDT ERROR:   File "\\rt60-06v\ansysem\v242\Win64\commonfiles\CPython\3_10\winx64\Release\python\lib\runpy.py", line 86, in _run_code
# PyAEDT ERROR:     exec(code, run_globals)
# PyAEDT ERROR:   File "C:\Users\User1\AppData\Roaming\.pyaedt_env\3_10\Scripts\ipython.exe\__main__.py", line 7, in <module>
# PyAEDT ERROR:     sys.exit(start_ipython())
# PyAEDT ERROR:   File "C:\Users\User1\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\IPython\__init__.py", line 130, in start_ipython
# PyAEDT ERROR:     return launch_new_instance(argv=argv, **kwargs)
# PyAEDT ERROR:   File "C:\Users\User1\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\traitlets\config\application.py", line 1075, in launch_instance
# PyAEDT ERROR:     app.start()
# PyAEDT ERROR:   File "C:\Users\User1\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\IPython\terminal\ipapp.py", line 317, in start
# PyAEDT ERROR:     self.shell.mainloop()
# PyAEDT ERROR:   File "<ipython-input-10-5794c4a5e292>", line 1, in <module>
# PyAEDT ERROR:     ex_obj_dict = circuit.excitation_objects
# PyAEDT ERROR:   File "C:\Users\User1\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ansys\aedt\core\application\analysis_nexxim.py", line 338, in excitation_objects
# PyAEDT ERROR:     for port in self.excitations:
# PyAEDT ERROR:   File "C:\Users\User1\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ansys\aedt\core\application\analysis_nexxim.py", line 324, in excitations
# PyAEDT ERROR:     ports = [p.replace("IPort@", "").split(";")[0] for p in self.modeler.oeditor.GetAllPorts() if "IPort@" in p]
# PyAEDT ERROR:   File "C:\Users\User1\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ansys\aedt\core\application\analysis_nexxim.py", line 226, in modeler
# PyAEDT ERROR:     self._modeler = ModelerNexxim(self)
# PyAEDT ERROR:   File "C:\Users\User1\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ansys\aedt\core\modeler\schematic.py", line 498, in __init__
# PyAEDT ERROR:     self._schematic = NexximComponents(self)
# PyAEDT ERROR:   File "C:\Users\User1\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ansys\aedt\core\modeler\circuits\primitives_nexxim.py", line 105, in __init__
# PyAEDT ERROR:     CircuitComponents.__init__(self, modeler)
# PyAEDT ERROR:   File "C:\Users\User1\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ansys\aedt\core\modeler\circuits\primitives_circuit.py", line 85, in __init__
# PyAEDT ERROR:     self.refresh_all_ids()
# PyAEDT ERROR:   File "C:\Users\User1\AppData\Roaming\.pyaedt_env\3_10\lib\site-packages\ansys\aedt\core\modeler\circuits\primitives_circuit.py", line 1072, in refresh_all_ids
# PyAEDT ERROR:     objID = int(o.schematic_id)
# PyAEDT ERROR: Invalid literal for int() with base 10: '5:1' on refresh_all_ids
# PyAEDT ERROR: Last Electronics Desktop Message - [error] invalid literal for int() with base 10: '5:1' on refresh_all_ids

# PyAEDT ERROR: **************************************************************
# PyAEDT INFO: ModelerNexxim class has been initialized!
# PyAEDT INFO: Modeler class has been initialized! Elapsed time: 0m 1sec

# The dictionary can be accessed anyhow:
ex_obj_dict
[... 206 lines cut ...]
tomli_w==1.2.0
tornado==6.4.1
traitlets==5.14.3
trame==3.7.1
trame-client==3.5.0
trame-server==3.2.3
trame-vtk==2.8.12
trame-vuetify==2.7.2
types-python-dateutil==2.9.0.20241003
typing_extensions==4.12.2
tzdata==2025.1
uri-template==1.3.0
urllib3==2.2.3
utm==0.7.0
vtk==9.3.1
wcwidth==0.2.13
webcolors==24.8.0
webencodings==0.5.1
websocket-client==1.8.0
websockets==14.1
widgetsnbextension==3.6.10
wslink==2.2.1
xarray==2024.10.0
xmltodict==0.14.2
yarl==1.18.3

FgPy93, 2025-03-06:
@Samuelopez-ansys

FgPy93, 2025-03-07:
Same applies for the excitations_by_type property.

[image]

Samuelopez-ansys, 2025-03-07:
Hi @FgPy93 ,

I am using the main and app.excitation_objects is working correctly. You are right about excitations_by_type, because it was used mainly in HFSS where you have multiple kind of excitations.

I have created a new Pull request to allow Circuit Interface Port excitations:

[image]

FgPy93, 2025-03-07:
@Samuelopez-ansys  thanks a lot for the fast response! 

As long as I just drag some ports into the project without any solution setup this is correct: 

[image]

When I add a simple Linear Network analysis like described and add for example some coaxes in between the ports, solve the project and retry the error occurs:

[image]

Samuelopez-ansys, 2025-03-07:
Got it @FgPy93 I think with this PR, the issue is fixed:

https://github.com/ansys/pyaedt/pull/5880

PR #5880 (2025-03-07): FEAT: Add excitations by type for Circuit interface ports
## Description
excitations_by_type was not covering Interface Ports in Circuit
When more than one port and component were included in the design, the returned excitation object was wrong. It is fixed with a correct Schematic ID assignment which is unique in the design.

## Issue linked
Close #5870 

## Checklist
- [ ] I have tested my changes locally.
- [ ] I have added necessary documentation or updated existing documentation.
- [ ] I have followed the coding style guidelines of this project.
- [ ] I have added appropriate tests (unit, [... cut]
closed 2025-03-07 by Samuelopez-ansys