ansys/pyaedt #5883: Bug located in get_antenna_data

Greesh03-Hon, 2025-03-07:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

There is a problem in get_antenna_data in pyaedt 15 version. I have different pyaedt 14 and 15 in two different PCs. The code which worked in pyaedt 14 is failing in pyaedt 15.

### Steps To Reproduce

[image]

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.12

### Installed packages

aiohappyeyeballs==2.4.3
aiohttp==3.11.8
aiosignal==1.3.1
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc4
ansys-tools-visualization-interface==0.6.0
anyio==4.4.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.1
async-lru==2.0.4
attrs==23.2.0
Babel==2.15.0
bcrypt==4.0.1
beautifulsoup4==4.12.3
bleach==6.1.0
blosc2==2.7.1
certifi==2021.10.8
cffi==1.16.0
chardet==4.0.0
charset-normalizer==2.0.12
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.2
contourpy==1.3.1
cryptography==41.0.5
cycler==0.12.1
debugpy==1.8.2
decorator==5.1.1
defusedxml==0.7.1
et_xmlfile==2.0.0
executing==2.0.1
fast_simplification==0.1.7
fastjsonschema==2.20.0
fiona==1.10.1
fonttools==4.55.0
fpdf==1.7.2
fpdf2==2.8.2
fqdn==1.5.1
frozenlist==1.5.0
ftd2xx==1.1.2
future==0.18.2
geopandas==0.14.4
gitdb==4.0.9
GitPython==3.1.27
grpcio==1.70.0
h11==0.14.0
httpcore==1.0.5
httpx==0.27.0
idna==3.3
imageio==2.36.1
intelhex==2.3.0
ipykernel==6.29.5
ipython==8.26.0
ipywidgets==8.1.5
iso8601==2.1.0
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.4
joblib==1.4.2
json5==0.9.25
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
jupyter-events==0.10.0
jupyter-lsp==2.2.5
jupyter_client==8.6.2
jupyter_core==5.7.2
jupyter_server==2.14.1
jupyter_server_terminals==0.5.3
jupyterlab==4.2.3
jupyterlab_pygments==0.3.0
jupyterlab_server==2.27.2
jupyterlab_widgets==3.0.13
kiwisolver==1.4.7
lxml==4.9.3
markdown-it-py==3.0.0
MarkupSafe==2.1.5
matlab==0.1
matplotlib==3.9.2
matplotlib-inline==0.1.7
mdurl==0.1.2
meshio==5.3.5
mistune==3.0.2
more-itertools==10.5.0
msgpack==1.1.0
multidict==6.1.0
nbclient==0.10.0
nbconvert==7.16.4
nbformat==5.10.4
ndindex==1.9.2
nest-asyncio==1.6.0
networkx==3.3
notebook==7.2.1
notebook_shim==0.2.4
numexpr==2.10.2
numpy==1.26.2
openpyxl==3.1.5
osmnx==1.9.4
overrides==7.7.0
packaging==24.1
pandas==2.2.3
pandocfilters==1.5.1
paramiko==3.3.1
parso==0.8.4
Pillow==10.1.0
platformdirs==4.2.2
plumbum==1.9.0
pooch==1.8.2
prometheus_client==0.20.0
prompt_toolkit==3.0.47
propcache==0.2.0
psutil==6.0.0
pure-eval==0.2.2
py-cpuinfo==9.0.0
pyadi-iio==0.0.16
pyaedt==0.14.1
pybrm_if @ file:///C:/Users/H569048/Downloads/90600936-REV-B/pybrm_if-1.3.3.tar.gz#sha256=56d4e393736d287bb2b27b28d2f9a452e2d77b000c1cd7715d3acccf3d3a2e93
pycparser==2.21
pydantic==2.9.2
pydantic_core==2.23.4
pyedb==0.31.0
Pygments==2.18.0
pykml==0.2.0
pylibiio==0.25
PyNaCl==1.5.0
pyparsing==3.2.0
pyproj==3.7.0
PyQRCode==1.2.1
pyserial==3.5
python-dateutil==2.9.0.post0
python-json-logger==2.0.7
pytz==2024.2
pyvista==0.44.2
pywin32==306
pywinpty==2.0.13
PyYAML==6.0.1
pyzmq==26.0.3
referencing==0.35.1
requests==2.32.3
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rich==13.9.4
rpds-py==0.19.0
rpyc==6.0.1
Rtree==1.3.0
scikit-learn==1.5.2
scikit-rf==1.4.1
scipy==1.14.1
scooby==0.10.0
Send2Trash==1.8.3
setuptools==70.3.0
shapely==2.0.6
six==1.16.0
smmap==5.0.0
sniffio==1.3.1
soupsieve==2.5
spur==0.3.23
SRTM.py==0.3.7
stack-data==0.6.3
tables==3.10.1
terminado==0.18.1
threadpoolctl==3.5.0
tinycss2==1.3.0
toml==0.10.2
tomli_w==1.2.0
tornado==6.4.1
traitlets==5.14.3
trame==3.7.0
trame-client==3.5.0
trame-server==3.2.3
trame-vtk==2.8.12
trame-vuetify==2.7.2
types-python-dateutil==2.9.0.20240316
typing_extensions==4.12.2
tzdata==2024.2
uri-template==1.3.0
urllib3==1.26.9
utm==0.7.0
vtk==9.3.1
wcwidth==0.2.13
webcolors==24.6.0
webencodings==0.5.1
websocket-client==1.8.0
websockets==14.1
widgetsnbextension==4.0.13
wslink==2.2.1
xlrd==2.0.1
yarl==1.18.0

Devin-Crawford, 2025-03-10:
@Greesh03-Hon : I'll take a look. A recent PR broke some things and this may be related.

Samuelopez-ansys, 2025-03-10:
Hi @Devin-Crawford , @Greesh03-Hon ,

Indeed in 0.15.0 there is a bug, in the last main, I think it is fixed. We just need to create a new release.

Greesh03-Hon, 2025-03-10:
Thanks for the swift resolve @Samuelopez-ansys , @Devin-Crawford . Please comment below whenever new release available to download 😊

closed 2025-03-27 by Samuelopez-ansys