ansys/pyaedt #5890: Trace do not show up after using import_trace() function into existing report in HFSS

FgPy93, 2025-03-10:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

When importing existing data from .csv file the added trace do not show up when trying to access the trace itself though the trace is visible in the GUI. Furthermore the trace in the GUI is also not editable. Workaround is to copy the trace in the GUI to make it editable.

### Steps To Reproduce

Described above.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.12

### Installed packages

aiohappyeyeballs==2.4.4
aiohttp==3.11.10
aiosignal==1.3.2
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc6
ansys-tools-visualization-interface==0.6.2
anyio==4.6.2.post1
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==2.4.1
async-lru==2.0.4
async-timeout==5.0.1
attrs==24.2.0
babel==2.16.0
beautifulsoup4==4.12.3
bleach==6.2.0
blosc2==2.7.1
certifi==2024.8.30
cffi==1.17.1
charset-normalizer==3.4.0
click==8.1.7
click-plugins==1.1.1
cligj==0.7.2
clr_loader==0.2.7.post0
colorama==0.4.6
comm==0.2.2
contourpy==1.3.0
cycler==0.12.1
debugpy==1.8.7
decorator==5.1.1
defusedxml==0.7.1
et_xmlfile==2.0.0
exceptiongroup==1.2.2
executing==2.1.0
fast_simplification==0.1.7
fastjsonschema==2.20.0
fiona==1.10.1
fonttools==4.55.4
fpdf2==2.8.2
fqdn==1.5.1
frozenlist==1.5.0
geopandas==0.14.4
gitdb==4.0.11
GitPython==3.1.43
grpcio==1.70.0
h11==0.14.0
httpcore==1.0.6
httpx==0.27.2
idna==3.10
imageio==2.36.0
ipycanvas==0.13.3
ipyevents==2.0.2
ipykernel==6.29.5
ipython==8.28.0
ipython-genutils==0.2.0
ipyvtklink==0.2.3
ipywidgets==7.8.5
isoduration==20.11.0
jedi==0.19.1
Jinja2==3.1.4
joblib==1.4.2
json5==0.9.25
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
jupyter-events==0.10.0
jupyter-lsp==2.2.5
jupyter_client==8.6.3
jupyter_core==5.7.2
jupyter_server==2.14.2
jupyter_server_terminals==0.5.3
jupyterlab==4.2.5
jupyterlab_pygments==0.3.0
jupyterlab_server==2.27.3
jupyterlab_widgets==1.1.11
kiwisolver==1.4.7
latex2mathml @ file:///S:/latex2mathml-3.77.0-py3-none-any.whl
lxml==5.3.0
markdown-it-py==3.0.0
MarkupSafe==3.0.2
matplotlib==3.9.2
matplotlib-inline==0.1.7
mdurl==0.1.2
meshio==5.3.5
mistune==3.0.2
more-itertools==10.5.0
msgpack==1.1.0
multidict==6.1.0
nbclient==0.10.0
nbconvert==7.16.4
nbformat==5.10.4
ndindex==1.9.2
nest-asyncio==1.6.0
networkx==3.3
notebook==7.2.2
notebook_shim==0.2.4
numexpr==2.10.1
numpy==2.2.2
openpyxl==3.1.5
osmnx==1.9.4
overrides==7.7.0
packaging==24.1
pandas==2.2.3
pandocfilters==1.5.1
parso==0.8.4
pillow==11.1.0
platformdirs==4.3.6
plumbum==1.9.0
pooch==1.8.2
prometheus_client==0.21.0
prompt_toolkit==3.0.48
propcache==0.2.1
psutil==6.1.1
pure_eval==0.2.3
py-cpuinfo==9.0.0
pyaedt==0.15.0
pycparser==2.22
pydantic==2.10.5
pydantic_core==2.27.2
pyedb==0.35.0
Pygments==2.18.0
pyparsing==3.2.0
pyproj==3.7.0
python-dateutil==2.9.0.post0
python-docx==1.1.2
python-json-logger==2.0.7
pytomlpp==1.0.13
pytz==2024.2
pyvista==0.44.1
pywin32==308
pywinpty==2.0.14
PyYAML==6.0.2
pyzmq==26.2.0
quantiphy==2.20
referencing==0.35.1
requests==2.32.3
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rich==13.9.3
# Editable install with no version control (roconnsim==0.1.0)
-e s:\f_grassl\pyaedt_lyon\automated-simulation\custom_data\roconnsim
ropylib @ file:///S:/F_Grassl/PyAEDT_Lyon/ropylib-0.3.16-py3-none-any.whl#sha256=ebd90a2f77104a18025059b937240fe173d5cab32e4140c00f4e04e74b323ee1
rpds-py==0.22.3
rpyc==6.0.1
Rtree==1.3.0
scikit-learn==1.5.2
scikit-rf==1.5.0
scipy==1.15.1
scooby==0.10.0
Send2Trash==1.8.3
shapely==2.0.6
six==1.17.0
smmap==5.0.1
sniffio==1.3.1
soupsieve==2.6
SRTM.py==0.3.7
stack-data==0.6.3
tables==3.10.1
terminado==0.18.1
threadpoolctl==3.5.0
tinycss2==1.4.0
toml==0.10.2
tomli==2.2.1
tomli_w==1.2.0
tornado==6.4.1
traitlets==5.14.3
trame==3.7.1
trame-client==3.5.0
trame-server==3.2.3
trame-vtk==2.8.12
trame-vuetify==2.7.2
types-python-dateutil==2.9.0.20241003
typing_extensions==4.12.2
tzdata==2025.1
uri-template==1.3.0
urllib3==2.2.3
utm==0.7.0
vtk==9.3.1
wcwidth==0.2.13
webcolors==24.8.0
webencodings==0.5.1
websocket-client==1.8.0
websockets==14.1
widgetsnbextension==3.6.10
wslink==2.2.1
xarray==2024.10.0
xmltodict==0.14.2
yarl==1.18.3

FgPy93, 2025-03-10:
@Samuelopez-ansys

Samuelopez-ansys, 2025-03-11:
Hi @FgPy93 , please could you explain the issue? 

I am testing the feature, but the imported trace is editable in the UI: https://github.com/ansys/pyaedt/blob/main/tests/system/visualization/test_12_1_PostProcessing.py#L355

Please try the test, and let me know what the issue is. I found one issue that I am fixing, related to the refresh of traces properties, which is not updated when a new trace is added

FgPy93, 2025-03-11:
@Samuelopez-ansys regarding the editability it was a fault on my side. Only the editable option was not set by standard after adding the trace. 

The "main" issue remains that an imported trace which came from external .csv data does not show up in the traces list of the report.

PR #5893 (2025-03-11): FIX: Update imported traces to trace property
## Description
Imported traces were not correctly updated.

## Issue linked
Close #5890 
Close #5889 

## Checklist
- [ ] I have tested my changes locally.
- [ ] I have added necessary documentation or updated existing documentation.
- [ ] I have followed the coding style guidelines of this project.
- [ ] I have added appropriate tests (unit, integration, system).
- [ ] I have reviewed my changes before submitting this pull request.
- [ ] I have linked the issue or issues that are solved by the PR if any.
- [ ] I have agreed with the Contributor Li [... cut]
closed 2025-03-11 by FgPy93