ansys/pyaedt #5958: I set location to volume or surface, but Icepak returns the same temperature value, when I use the method get_temperature_extremum.

chs513, 2025-03-26:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

I set location to volume or surface, but Icepak returns the same temperature value, when I use the method get_temperature_extremum.

### Steps To Reproduce

```
from ansys.aedt.core import Icepak

ipk = Icepak()

temp = ipk.post.get_temperature_extremum(
    assignment='Box2',
    max_min='Max',
    # location='Volume'
    location='Surface'
)

print(temp)
```

[test_result.zip](https://github.com/user-attachments/files/19459324/test_result.zip)

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

annotated-types==0.7.0
ansys-pythonnet==3.1.0rc6
appdirs==1.4.4
argcomplete==3.6.0
asttokens==3.0.0
attrs==24.2.0
autopep8==2.3.2
bcrypt==4.3.0
certifi==2025.1.31
cffi==1.17.1
chardet==4.0.0
charset-normalizer==3.4.1
click==8.1.8
clr_loader==0.2.7.post0
colorama==0.4.6
contourpy==1.3.1
cryptography==44.0.2
cycler==0.12.1
decorator==5.1.1
defusedxml==0.7.1
epc==0.0.5
exceptiongroup==1.2.2
executing==2.2.0
fonttools==4.55.8
fpdf2==2.8.2
grpcio==1.70.0
idna==2.10
ipython==8.32.0
jedi==0.19.2
Jinja2==3.0.1
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
kiwisolver==1.4.8
Mako==1.3.9
MarkupSafe==3.0.2
matplotlib==3.10.0
matplotlib-inline==0.1.7
numpy==2.2.2
orjson==3.10.15
packaging==24.2
pandas==2.2.3
paramiko==3.5.1
parso==0.8.4
Pillow==8.4.0
pipx==1.7.1
platformdirs==4.3.6
plumbum==1.9.0
pooch==1.8.2
prompt_toolkit==3.0.50
psutil==6.1.1
pure_eval==0.2.3
pyaedt==0.15.2
pycodestyle==2.12.1
pycparser==2.22
pydantic==2.10.6
pydantic_core==2.27.2
pyedb==0.37.0
pygame==2.6.1
Pygments==2.19.1
pygubu==0.36.3
pygubu-designer==0.40.2
PyNaCl==1.5.0
pyparsing==3.2.1
PyQt6==6.4.2
pyqt6-plugins==6.4.2.2.3
PyQt6-Qt6==6.4.3
pyqt6-tools==6.4.2.3.3
PyQt6_sip==13.10.0
python-dateutil==2.9.0.post0
python-dotenv==1.0.1
pythonnet==3.0.5
pytz==2025.1
pyvista==0.44.2
pywin32==308
PyYAML==6.0.2
qt6-applications==6.4.3.2.3
qt6-tools==6.4.3.1.3
RapidFuzz==3.12.2
referencing==0.35.1
requests==2.25.1
rpds-py==0.22.3
rpyc==6.0.1
Rtree==1.3.0
scikit-rf==1.5.0
scipy==1.15.1
scooby==0.10.0
screeninfo==0.8.1
sexpdata==1.0.2
six==1.17.0
stack-data==0.6.3
tkdesigner==1.0.7
toml==0.10.2
tomli==2.2.1
tomli_w==1.2.0
traitlets==5.14.3
typing_extensions==4.12.2
tzdata==2025.1
urllib3==1.26.20
userpath==1.9.2
vtk==9.3.1
watchdog==6.0.0
wcwidth==0.2.13

lorenzovecchietti, 2025-03-28:
Not a bug. 
As you see the calculator output is the same for both the Volume and Surface query:
[image]
[image]

closed 2025-03-28 by lorenzovecchietti

chs513, 2025-03-28:
@lorenzovecchietti According to common sense, these two temperature values for both the Volume and Surface should be different.

lorenzovecchietti, 2025-03-28:
You're right but that might be a problem with the solution setup or with the mesh. Not with pyAEDT. If you think there is a bug in AEDT or you have some doubt in the setup of the simulation, please open a support ticket in the customer portal.

chs513, 2025-03-28:
@lorenzovecchietti  Got it. Thanks!