ansys/pyaedt #5961: Add a getting the junction temperature method for Icepak when post processing.

chs513, 2025-03-26:
### Description of the feature

Add a getting the junction temperature method for Icepak when post processing.

lorenzovecchietti, 2025-03-28:
You can do that with the field summary class and methods.

closed 2025-03-28 by lorenzovecchietti

lorenzovecchietti, 2025-03-28:
https://aedt.docs.pyansys.com/version/stable/API/visualization/_autosummary/ansys.aedt.core.visualization.post.field_summary.FieldSummary.html#ansys.aedt.core.visualization.post.field_summary.FieldSummary

If you have to create a field summary, add your network boundary with `add_calculation`, and then you can get internal node temperature.

If you need and example of the field summary class usare, you can find it here.

chs513, 2025-03-31:
@lorenzovecchietti Could you give me an example about the field summary and the add_calculation?

lorenzovecchietti, 2025-03-31:
Sure! [Here](https://examples.aedt.docs.pyansys.com/version/dev/examples/electrothermal/graphic_card.html#Postprocess)

You have to create a field summary, add your network boundary with add_calculation, and then you can get internal node temperature.

chs513, 2025-03-31:
@lorenzovecchietti Can you help me point out what is wrong?
```
from ansys.aedt.core import Icepak

ipk = Icepak()

temperature_fs = ipk.post.create_field_summary()
temperature_fs.add_calculation(
    entity="Boundary",
    geometry="Volume",
    geometry_name='Box3',
    quantity="Temperature"
)
temperature_fs = temperature_fs.get_field_summary_data(pandas_output=True)
temperature_fs.head()
```

[image]
[test_result.zip](https://github.com/user-attachments/files/19531323/test_result.zip)

Samuelopez-ansys, 2025-03-31:
I think you are mixing single quote and double:
'box3' 

"box3"

chs513, 2025-03-31:
Hi @Samuelopez-ansys, 'Box3' -> "Box3", the same error.

lorenzovecchietti, 2025-03-31:
It seems the argument name might be a bit confusing; it should be "entity_name".  When selecting the "boundary" entity, "geometry_name" should correspond to the boundary name that you see under the thermal three.

chs513, 2025-03-31:
@lorenzovecchietti I modified the "geometry_name" parameter, but I got the same error. Could you show me right codes directly?
```
from ansys.aedt.core import Icepak

ipk = Icepak()

temperature_fs = ipk.post.create_field_summary()
temperature_fs.add_calculation(
    entity="Boundary",
    geometry="Volume",
    # geometry_name="Box3",
    geometry_name="Network1",
    quantity="Temperature"
)
temperature_fs = temperature_fs.get_field_summary_data(pandas_output=True)
temperature_fs.head()
```