ansys/pyaedt #5968: Use pyaedt inside a Docker Container

ptrem, 2025-03-26:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

Hey,
I would like to know if it is possible to use pyaedt inside a Docker Container?
The file could be something simple like:

```
from ansys.aedt.core import Hfss

hfss = Hfss(version="2023.2",
            non_graphical=True)

origin = [0, 0, 0]
dimensions = [10, 5, 20]
box = hfss.modeler.create_box(
    origin=origin,
    dimensions=dimensions,
    name="mybox",
    material="copper"
)
```

### Steps To Reproduce

So far I have tried some different variants I've set the environmental variables in the dockerfile and defined the licence server :
```
export ANSYSEM_ROOT222=/path/to/AedtRoot/AnsysEM/v222/Linux64
export LD_LIBRARY_PATH=$ANSYSEM_ROOT222/common/mono/Linux64/lib64:$ANSYSEM_ROOT222/Delcross:$LD_LIBRARY_PATH
```
At the end I got the following error. I also tried to define the port 
```
ANSYSEM_ROOT232: /opt/AnsysEM/v232/Linux64
Attempting to connect on port 50421...
PyAEDT INFO: Python version 3.10.12 (main, Feb  4 2025, 14:57:36) [GCC 11.4.0].
PyAEDT INFO: PyAEDT version 0.15.2.
PyAEDT INFO: Initializing new Desktop session.
PyAEDT INFO: Log on console is enabled.
PyAEDT INFO: Log on file /tmp/pyaedt_root_f9cabee9-5a07-4334-9f7c-777de892593d.log is enabled.
PyAEDT INFO: Log on AEDT is disabled.
PyAEDT INFO: Debug logger is disabled. PyAEDT methods will not be logged.
PyAEDT INFO: Launching PyAEDT with gRPC plugin.
PyAEDT INFO: New AEDT session is starting on gRPC port 50421.
PyAEDT WARNING: Failed to start on gRPC port: 50421. Trying to start on 58555.
PyAEDT ERROR: Failed to start on gRPC port: 58555.
```

Maybe anyone has already successfully run a pyaedt file inside a docker container.
Thanks for your help.

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.10

### Installed packages

pyaedt

ptrem, 2025-03-27:
I found a solution by first starting AEDT in listing mode on the remote machine (as described in the pyAEDT [Documentation](https://aedt.docs.pyansys.com/version/stable/Getting_started/ClientServer.html)),

I first add the display by running:
`export DISPLAY=:1`

and start AEDT with:
`/opt/AnsysEM/v232/Linux64/ansysedt -grpcsrv 50051`

This should make it possible to make AEDT accessible out of a Docker container.

I also provide a solution to automate the file execution in a Docker container.

### Example Python Script

`run_pyaedt.py`

```
from ansys.aedt.core import Hfss

# Launch AEDT (replace the version with the one you have licensed)
hfss = Hfss(
    project="MyProject",
    design="MyDesign",
    version="2023.2",
    port=50051,
    non_graphical=False,
    new_desktop=False)

# Create a simple box model
origin = [0, 0, 0]
sizes = [10, 5, 20]
box = hfss.modeler.create_box(origin=origin, sizes=sizes, name="mybox", material="copper")

hfss.release_desktop()
```

### Dockerfile

`dockerfile`

```
FROM python:3.12-slim

# Set environment variables to suppress interactive prompts during installation
ENV DEBIAN_FRONTEND=noninteractive

# Set environment variables for AEDT installation.
# Adjust these paths to match your AEDT installation.
ENV LD_LIBRARY_PATH=""
ENV ANSYSEM_ROOT232=/opt/AnsysEM/v232/Linux64
ENV LD_LIBRARY_PATH=${ANSYSEM_ROOT232}/common/mono/Linux64/lib64:${ANSYSEM_ROOT232}/Delcross:$LD_LIBRARY_PATH

# Install PyAEDT from PyPI
RUN pip3 install pyaedt pyvista

# Copy your PyAEDT script into the image
COPY run_pyaedt.py /opt/run_pyaedt.py

CMD ["python3", "/opt/run_pyaedt.py"]
```

### Startup File

`run.sh`

```
#!/bin/bash

# Define variables for reuse
GRPC_PORT=50051
PYAEDT_PATH="/opt/AnsysEM/v232/Linux64"

echo "Set Display to :1"
export DISPLAY=:1

# Start AEDT gRPC server on host
echo "Starting AEDT gRPC server on host..."
$PYAEDT_PATH/ansysedt -grpcsrv $GRPC_PORT &
AEDT_PID=$!

# Wait for gRPC server to initialize
sleep 80
echo "Checking if AEDT gRPC server is running..."
if ps -p $AEDT_PID > /dev/null; then
    echo "AEDT gRPC server started successfully with PID $AEDT_PID"
else
    echo "Failed to start AEDT gRPC server"
    kill $XVFB_PID
    exit 1
fi

# Get host IP address for container to use
HOST_IP=$(hostname -I | awk '{print $1}')
echo "Host IP: $HOST_IP"

# Build Docker image if needed
echo "Building Docker container..."
docker build -t pyaedt-client .

# Run Docker container with connection to host
echo "Running PyAEDT client in Docker container..."
docker run --rm -it --network=host -p $GRPC_PORT:$GRPC_PORT -v $PYAEDT_PATH:$PYAEDT_PATH pyaedt-client
```

closed 2025-03-27 by ptrem

SMoraisAnsys, 2025-03-27:
Hey @ptrem, I'm glad you found a solution that suits your needs. Also, thanks for sharing your feedback to the community !

jrbe228, 2026-01-21:
Is there now an officially maintained Dockerfile for AEDT?  
Similar Dockerfiles for [MAPDL](https://github.com/ansys/pymapdl/blob/main/docker/build/Dockerfile) and [Fluent](https://github.com/ansys/pyfluent/blob/main/docker/fluent_261/Dockerfile) have been very helpful.