ansys/pyaedt #5971: Bug located in post_common_3d.py

sara-tell, 2025-03-27:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

I'm using pyaedt 0.15.1 which is supposed to be supported for python 3.8. I'm using python 3.8.3. With python 3.8, PostProcessor3D class is not working since adding get_field_extremum in commit 5840. This class method uses dict[str, str] type hints, which was introduced in python 3.9. 

### Steps To Reproduce

hfss = Hfss()
hfss.post

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.8

### Installed packages

annotated-types==0.7.0
ansys-api-edb==1.0.10
ansys-edb-core==0.1.9
ansys-pythonnet==3.1.0rc6
attrs==24.2.0
certifi==2025.1.31
cffi==1.17.1
charset-normalizer==3.4.1
clr_loader==0.2.7.post0
contourpy==1.1.1
cycler==0.12.1
defusedxml==0.7.1
distro==1.9.0
dotnetcore2==3.1.23
fonttools==4.56.0
fpdf2==2.8.2
grpcio==1.70.0
idna==3.10
importlib_resources==6.4.5
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
kiwisolver==1.4.7
matplotlib==3.7.5
numpy==1.24.4
packaging==24.2
pandas==2.0.3
pillow==10.4.0
pkgutil_resolve_name==1.3.10
platformdirs==4.3.6
plumbum==1.9.0
pooch==1.8.2
protobuf==4.25.6
psutil==7.0.0
pyaedt==0.15.1
pycparser==2.22
pydantic==2.10.6
pydantic_core==2.27.2
pyedb==0.39.1
pyparsing==3.1.4
python-dateutil==2.9.0.post0
pytz==2025.2
pyvista==0.44.2
PyYAML==6.0.2
referencing==0.35.1
requests==2.32.3
rpds-py==0.20.1
rpyc==6.0.1
Rtree==1.3.0
scikit-rf==1.6.2
scipy==1.10.1
scooby==0.10.0
six==1.17.0
toml==0.10.2
tomli==2.2.1
tomli_w==1.0.0
typing_extensions==4.12.2
tzdata==2025.2
urllib3==2.2.3
vtk==9.3.1
zipp==3.20.2

Samuelopez-ansys, 2025-03-27:
Hi @sara-tell ,

Thank you for the catch, we will fix it, but please consider moving to Python 3.10 at least. 

Python 3.8 is end-of-life

https://devguide.python.org/versions/

sara-tell, 2025-03-27:
I'm using a linux server where I have no control over the installed versions, otherwise I would've upgraded, but I will suggest to the maintainers to do so

PR #5972 (2025-03-27): FIX: Compatibility with Python 3.8
## Description
**Please provide a brief description of the changes made in this pull request.**

## Issue linked
Close #5971 

## Checklist
- [ ] I have tested my changes locally.
- [ ] I have added necessary documentation or updated existing documentation.
- [ ] I have followed the coding style guidelines of this project.
- [ ] I have added appropriate tests (unit, integration, system).
- [ ] I have reviewed my changes before submitting this pull request.
- [ ] I have linked the issue or issues that are solved by the PR if any.
- [ ] I have agreed with the C [... cut]
closed 2025-03-27 by Samuelopez-ansys