ansys/pyaedt #5975: PYAEDT Extract Delphi Network

bbt6750, 2025-03-27:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

AEDT does not recognize that PyAEDT was installed when trying to use Extract Delphi Network Toolkit

Error message displayed is:
Win32Exception: WindowsError  In file "C:/Program Files/AnsysEM/v242/Win64/syslib/Toolkits/Icepak/Modeling/IC Packages/Extract_Delphi_Network.py", line 85 ---- While executing script: C:/Program Files/AnsysEM/v242/Win64/syslib/Toolkits/Icepak/Modeling/IC Packages/Extract_Delphi_Network.py

### Steps To Reproduce

- - Install PYAEDT using wheelhouse methodInstall PYAEDT using wheelhouse method
- - Attempt to run "Extract Delphi Network" toolkit from Automation tab or try to run script for Extract_Delphi_Network.py

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

2to3==1.0
aiohappyeyeballs==2.6.1
aiohttp==3.11.13
aiosignal==1.3.2
annotated-types==0.7.0
ansys-api-fluent==0.3.22
ansys-api-platform-instancemanagement==1.0.0
ansys-fluent-core==0.20.dev12
ansys-platform-instancemanagement==1.1.2
ansys-pythonnet==3.1.0rc6
ansys-tools-visualization-interface==0.8.3
ansys-units==0.3.2
anyio==4.8.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asttokens==3.0.0
async-lru==2.0.4
async-timeout==5.0.1
attrs==23.2.0
babel==2.17.0
beartype==0.17.2
beautifulsoup4==4.13.3
bleach==6.2.0
blosc2==2.7.1
certifi==2023.11.17
cffi==1.16.0
charset-normalizer==3.3.2
clr-loader==0.2.6
colorama==0.4.6
comm==0.2.2
contourpy==1.3.1
cycler==0.12.1
debugpy==1.8.13
decorator==5.2.1
defusedxml==0.7.1
distro==1.9.0
docker==6.1.3
et_xmlfile==2.0.0
exceptiongroup==1.2.0
execnet==2.0.2
executing==2.2.0
fast_simplification==0.1.9
fastjsonschema==2.21.1
fonttools==4.50.0
fpdf2==2.8.2
fqdn==1.5.1
frozenlist==1.5.0
future==0.18.0
geopandas==1.0.1
googleapis-common-protos==1.52.0
grpcio==1.70.0
grpcio-health-checking==1.48.1
grpcio-status==1.26.0
h11==0.14.0
h5py==3.10.0
httpcore==1.0.7
httpx==0.28.1
idna==3.6
imageio==2.37.0
importlib_metadata==7.1.0
iniconfig==2.0.0
ipycanvas==0.13.3
ipyevents==2.0.2
ipykernel==6.29.5
ipython==8.34.0
ipython-genutils==0.2.0
ipyvtklink==0.2.3
ipywidgets==7.8.5
isoduration==20.11.0
jedi==0.19.2
Jinja2==3.1.6
joblib==1.4.2
json5==0.10.0
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
jupyter-events==0.12.0
jupyter-lsp==2.2.5
jupyter_client==8.6.3
jupyter_core==5.7.2
jupyter_server==2.15.0
jupyter_server_terminals==0.5.3
jupyterlab==4.3.5
jupyterlab_pygments==0.3.0
jupyterlab_server==2.27.3
jupyterlab_widgets==1.1.11
kiwisolver==1.4.5
llvmlite==0.39.1
lxml==4.9.4
markdown-it-py==3.0.0
MarkupSafe==3.0.2
matplotlib==3.6.0
matplotlib-inline==0.1.7
mdurl==0.1.2
meshio==5.3.5
mistune==3.1.2
more-itertools==10.6.0
msgpack==1.1.0
multidict==6.1.0
nbclient==0.10.2
nbconvert==7.16.6
nbformat==5.10.4
ndindex==1.9.2
nest-asyncio==1.6.0
networkx==3.4.2
notebook==7.3.2
notebook_shim==0.2.4
numba==0.56.4
numexpr==2.10.2
numpy==2.2.3
openpyxl==3.1.5
osmnx==2.0.1
overrides==7.7.0
packaging==24.0
pandas==2.0.3
pandocfilters==1.5.1
parso==0.8.4
pillow==10.2.0
pip-system-certs==4.0
platformdirs==3.11.0
pluggy==1.4.0
plumbum==1.9.0
pooch==1.8.2
prometheus_client==0.21.1
prompt_toolkit==3.0.50
propcache==0.3.0
protobuf==3.19.3
psutil==5.9.2
pure_eval==0.2.3
py==1.11.0
py-cpuinfo==9.0.0
pyaedt==0.15.1
pybind11==2.10.0
pycparser==2.21
pydantic==2.10.6
pydantic_core==2.27.2
pyedb==0.38.0
Pygments==2.19.1
Pymem==1.13.1
pyogrio==0.10.0
pyparsing==3.1.2
pyproj==3.7.1
pytest==7.2.1
pytest-forked==1.6.0
pytest-xdist==1.31.0
python-dateutil==2.9.0.post0
python-json-logger==3.3.0
pytz==2024.1
pyvista==0.44.2
pywin32==304
pywinpty==2.0.15
PyYAML==6.0.1
pyzmq==26.3.0
referencing==0.35.1
requests==2.31.0
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rich==13.9.4
rpds-py==0.23.1
rpyc==6.0.1
rtree==1.4.0
scikit-learn==1.6.1
scikit-rf==1.6.2
scipy==1.15.2
SCons==4.0.1
scooby==0.10.0
Send2Trash==1.8.3
shapely==2.0.2
sip==6.5.1
six==1.16.0
sniffio==1.3.1
soupsieve==2.6
stack-data==0.6.3
tables==3.10.1
terminado==0.18.1
threadpoolctl==3.5.0
tinycss2==1.4.0
toml==0.10.2
tomli==2.0.1
tomli_w==1.2.0
tornado==6.4.2
traitlets==5.14.3
trame==3.8.1
trame-client==3.6.0
trame-server==3.4.0
trame-vtk==2.8.15
trame-vuetify==2.8.1
types-python-dateutil==2.9.0.20241206
typing_extensions==4.12.2
tzdata==2024.1
uri-template==1.3.0
urllib3==1.26.10
vtk==9.3.1
wcwidth==0.2.13
webcolors==24.11.1
webencodings==0.5.1
websocket-client==1.7.0
websockets==15.0.1
widgetsnbextension==3.6.10
wrapt==1.16.0
wslink==2.3.2
yarl==1.18.3
zipp==3.18.1

lorenzovecchietti, 2025-03-28:
Hi @bbt6750, as this issue appears to be related to the product itself rather than pyAEDT, could you please submit a support ticket through the Ansys Support portal? If you have already done so, please provide the ticket number so I can escalate it to the appropriate support team.

closed 2025-03-28 by lorenzovecchietti

bbt6750, 2025-03-28:
Hi Lorenzo,

I had submitted two tickets previously and both were declined as not officially supported through the Support Portal.  Joey Du (cc’d here) had been helping me with some steps but recommended I submit my inquiry here.  My last ANSYS support ticket was Case #00158127.  Please let me know who can help with this installation issue.

Best Regards,
Blake Travis

ZjQcmQRYFpfptBannerStart
This message was sent from outside of Texas Instruments.

Do not click links or open attachments unless you recognize the source of this email and know the content is safe.

    Report Suspicious  <https://us-phishalarm-ewt.proofpoint.com/EWT/v1/G3vK!urdgNt_mXg7zIyxOllsAN1SsQy-KbNHQGaJuxKJOXS41g7C4L6MEEV4vpn2dKvUrK5Txmuinrao85puSPvMi3T-7Dr-Dyg$>   ‌

ZjQcmQRYFpfptBannerEnd