ansys/pyaedt #5984: [Regarding LTspice Import Function].be able to recognize custom device models

hausen001, 2025-03-31:
### Description of the feature

Hi Expert,
I hope this message finds you well.
As server applications such as AI ,HPC have higher power requirements applications are becoming increasingly important in our projects, I’m currently evaluating the LTspice import function to better support simulation needs.
I have been testing the LTspice import functionality and found that standard components such as resistors, inductors, and capacitors (R, L, C) are successfully imported. 
However, I’ve encountered issues when working with customized circuit symbols — they do not seem to import correctly or as expected.
Based on the results so far (please refer to the screenshots attached), I’m wondering if there are specific steps or considerations I might have overlooked during the import process.
Would you be able to advise if there are additional settings, symbol definitions, or constraints I should be aware of when importing custom components from LTspice?
Thank you very much for your support.

Samuelopez-ansys, 2025-04-01:
Hi @hausen001 ,

The LTSPice converter is for now only converting RLC and sources:

https://github.com/ansys/pyaedt/blob/main/src/ansys/aedt/core/circuit.py#L2440

As you can see in the method, you can add a config file which allows to import additional kind of devices (if they exist in Circuit), here is an example:

https://github.com/ansys/pyaedt/blob/main/src/ansys/aedt/core/misc/asc_circuit_mapping.json

Please see the implementation, and if you miss something, feel free to contribute :)

hausen001, 2025-04-11:
[like]  Bo Yang Tsai reacted to your message:

[External Sender]

https://github.com/ansys/pyaedt/blob/main/src/ansys/aedt/core/circuit.py#L2440

https://github.com/ansys/pyaedt/blob/main/src/ansys/aedt/core/misc/asc_circuit_mapping.json

https://github.com/ansys/pyaedt/blob/main/src/ansys/aedt/core/circuit.py#L2440

https://github.com/ansys/pyaedt/blob/main/src/ansys/aedt/core/misc/asc_circuit_mapping.json

closed 2025-04-28 by Samuelopez-ansys