ansys/pyaedt #5988: The python file execute normally, but there are many errors when the exe file executes.

chs513, 2025-04-01:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

The python file execute normally, and I convert it to an .exe file by **pyinstaller**. There are many errors when the .exe file executes, for example "Error occurred while coping file".

### Steps To Reproduce

```
from ansys.aedt.core import Icepak
import csv

ipk = Icepak()

box = ipk.modeler.create_box([0, 0, 0], [5, 5, 1])

ipk.assign_solid_block(object_name=box.name, power_assignment='0.1W')

for face in ipk.modeler['Region'].faces:
    ipk.assign_free_opening(
        assignment=face.id,
        flow_type='Pressure'
    )

ipk.mesh.generate_mesh()

setup1 = ipk.create_setup()
setup1.props['Flow Regime'] = 'Laminar'
setup1.props['Convergence Criteria - Max Iterations'] = '100'
setup1.props['Convergence Criteria - Flow'] = '0.001'
setup1.props['Radiation Model'] = 'Discrete Ordinates Model'
setup1.props['Flow Iteration Per Radiation Iteration'] = '10'
setup1.props['ThetaDivision'] = '3'
setup1.props['PhiDivision'] = '3'
setup1.props['ThetaPixels'] = '3'
setup1.props['PhiPixels'] = '3'
setup1.props['Include Gravity'] = True

ipk.analyze()

with open(r'D:\test.csv', 'w', newline='') as f:
    writer = csv.writer(f)
    writer.writerow(['componentName', 'MaxTemperature'])
    maxTemp = ipk.post.get_temperature_extremum(
        assignment=box.name,
        max_min='Max',
        location='Volume'
    )
    maxTemp = round(maxTemp[1], 1)
    writer.writerow([box.name, maxTemp])
```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

altgraph==0.17.4
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc6
appdirs==1.4.4
argcomplete==3.6.0
asttokens==3.0.0
attrs==24.2.0
autopep8==2.3.2
bcrypt==4.3.0
certifi==2025.1.31
cffi==1.17.1
chardet==4.0.0
charset-normalizer==3.4.1
click==8.1.8
clr_loader==0.2.7.post0
colorama==0.4.6
contourpy==1.3.1
cryptography==44.0.2
cycler==0.12.1
decorator==5.1.1
defusedxml==0.7.1
epc==0.0.5
exceptiongroup==1.2.2
executing==2.2.0
fonttools==4.55.8
fpdf2==2.8.2
grpcio==1.70.0
idna==2.10
ipython==8.32.0
jedi==0.19.2
Jinja2==3.0.1
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
kiwisolver==1.4.8
Mako==1.3.9
MarkupSafe==3.0.2
matplotlib==3.10.0
matplotlib-inline==0.1.7
numpy==2.2.2
orjson==3.10.15
packaging==24.2
pandas==2.2.3
paramiko==3.5.1
parso==0.8.4
pefile==2023.2.7
Pillow==8.4.0
pipx==1.7.1
platformdirs==4.3.6
plumbum==1.9.0
pooch==1.8.2
prompt_toolkit==3.0.50
psutil==6.1.1
pure_eval==0.2.3
pyaedt==0.15.3
pycodestyle==2.12.1
pycparser==2.22
pydantic==2.10.6
pydantic_core==2.27.2
pyedb==0.37.0
pygame==2.6.1
Pygments==2.19.1
pygubu==0.36.3
pygubu-designer==0.40.2
pyinstaller==6.12.0
pyinstaller-hooks-contrib==2025.2
PyNaCl==1.5.0
pyparsing==3.2.1
PyQt6==6.4.2
pyqt6-plugins==6.4.2.2.3
PyQt6-Qt6==6.4.3
pyqt6-tools==6.4.2.3.3
PyQt6_sip==13.10.0
python-dateutil==2.9.0.post0
python-dotenv==1.0.1
pythonnet==3.0.5
pytz==2025.1
pyvista==0.44.2
pywin32==308
pywin32-ctypes==0.2.3
PyYAML==6.0.2
qt6-applications==6.4.3.2.3
qt6-tools==6.4.3.1.3
RapidFuzz==3.12.2
referencing==0.35.1
requests==2.25.1
rpds-py==0.22.3
rpyc==6.0.1
Rtree==1.3.0
scikit-rf==1.5.0
scipy==1.15.1
scooby==0.10.0
screeninfo==0.8.1
sexpdata==1.0.2
six==1.17.0
stack-data==0.6.3
tkdesigner==1.0.7
toml==0.10.2
tomli==2.2.1
tomli_w==1.2.0
traitlets==5.14.3
typing_extensions==4.12.2
tzdata==2025.1
urllib3==1.26.20
userpath==1.9.2
vtk==9.3.1
watchdog==6.0.0
wcwidth==0.2.13

lorenzovecchietti, 2025-04-02:
Do you encounter any issue in running the native code? If yes, what? If no, is there any reason why you think this is a pyAEDT issue?
Are you sure this is not an [expected issue of using pyinstaller](https://pyinstaller.org/en/stable/common-issues-and-pitfalls.html)?

chs513, 2025-04-03:
@lorenzovecchietti No issue in running the native code. Sorry, I'm not familiar with pyinstaller.

closed 2025-04-03 by chs513

lorenzovecchietti, 2025-04-03:
No problem. What I would check is if the initialization of AEDT works properly with pyinstaller. But bear in mind that I'm absoluterly not familiar too.

Also, if your need is just to make the script clickable you can consider:
- to integrate the call af the python script in some powershell script
- to add an extension inside aedt (if you installed pyaedt from the application)
[image]

chs513, 2025-04-03:
@lorenzovecchietti Thanks for your reply.