ansys/pyaedt #5991: Bug located in assign_balloon() for 2d electrostatic balloon type

Beanhum, 2025-04-01:
### Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### Description of the bug

For the Maxwell 2D, balloon boundary **assign_balloon(assignment, boundary=None)** only has two parameters, but if I use the 2D Electrostatic simulation, I have to choose balloon boundary type between Voltage and Charge:

[image]

What should I do to put that type into the **assign_balloon()**?
Thanks a lot for your assistance!

### Steps To Reproduce

Maxwell 2D Electrostatic 
**assign_balloon(assignment, boundary=None)**
balloon type

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.11

### Installed packages

aiohappyeyeballs==2.4.4
aiohttp==3.11.11
aiosignal==1.3.2
annotated-types==0.7.0
ansys-pythonnet==3.1.0rc5
ansys-tools-visualization-interface==0.6.2
anyio==4.8.0
asttokens==3.0.0
attrs==24.2.0
blosc2==3.0.0
certifi==2024.12.14
cffi==1.17.1
charset-normalizer==3.4.1
clr_loader==0.2.7.post0
colorama==0.4.6
comm==0.2.2
contourpy==1.3.1
cycler==0.12.1
debugpy==1.8.11
decorator==5.1.1
defusedxml==0.7.1
et_xmlfile==2.0.0
executing==2.1.0
fast_simplification==0.1.9
fonttools==4.55.3
fpdf2==2.8.2
frozenlist==1.5.0
geopandas==1.0.1
grpcio==1.70.0
h11==0.14.0
httpcore==1.0.7
httpx==0.28.1
idna==3.10
imageio==2.36.1
ipykernel==6.29.5
ipython==8.31.0
jedi==0.19.2
joblib==1.4.2
jsonschema==4.23.0
jsonschema-specifications==2024.10.1
jupyter_client==8.6.3
jupyter_core==5.7.2
kiwisolver==1.4.8
markdown-it-py==3.0.0
matplotlib==3.9.4
matplotlib-inline==0.1.7
mdurl==0.1.2
meshio==5.3.5
more-itertools==10.5.0
msgpack==1.1.0
multidict==6.1.0
ndindex==1.9.2
nest-asyncio==1.6.0
networkx==3.4.2
numexpr==2.10.2
numpy==2.2.1
openpyxl==3.1.5
osmnx==2.0.1
packaging==24.2
pandas==2.2.3
parso==0.8.4
pillow==11.1.0
platformdirs==4.3.6
plumbum==1.9.0
pooch==1.8.2
prompt_toolkit==3.0.48
propcache==0.2.1
psutil==6.1.1
pure_eval==0.2.3
py-cpuinfo==9.0.0
pyaedt==0.15.2
pycparser==2.22
pydantic==2.9.2
pydantic_core==2.23.4
pyedb==0.34.3
Pygments==2.19.1
pyogrio==0.10.0
pyparsing==3.2.1
pyproj==3.7.0
python-dateutil==2.9.0.post0
pytz==2024.2
pyvista==0.44.2
pywin32==308
PyYAML==6.0.2
pyzmq==26.2.0
referencing==0.35.1
requests==2.32.3
rich==13.9.4
rpds-py==0.22.3
rpyc==6.0.1
Rtree==1.3.0
scikit-learn==1.6.0
scikit-rf==1.5.0
scipy==1.15.0
scooby==0.10.0
shapely==2.0.6
six==1.17.0
sniffio==1.3.1
SRTM.py==0.3.7
stack-data==0.6.3
tables==3.10.2
threadpoolctl==3.5.0
toml==0.10.2
tomli_w==1.1.0
tornado==6.4.2
traitlets==5.14.3
trame==3.7.6
trame-client==3.5.1
trame-server==3.2.7
trame-vtk==2.8.14
trame-vuetify==2.8.0
typing_extensions==4.12.2
tzdata==2024.2
urllib3==2.3.0
utm==0.7.0
vtk==9.3.1
wcwidth==0.2.13
websockets==14.1
wslink==2.2.2
yarl==1.18.3

anur7, 2025-04-04:
Hi @Beanhum. I added an argument to choose between voltage or charge balloon type boundary for electrostatic solver. Please take a look and review this pull request #6011

closed 2025-04-07 by anur7

anur7, 2025-04-07:
Hi @Beanhum. The PR #6011 has been merged, you can now create balloon boundary as voltage or charge type for Maxwell electrostatic solver.

Beanhum, 2025-04-10:
> Hi [@Beanhum](https://github.com/Beanhum). The PR [#6011](https://github.com/ansys/pyaedt/pull/6011) has been merged, you can now create balloon boundary as voltage or charge type for Maxwell electrostatic solver.

Hi anur, thanks for your help!  I see the latest version is v0.15.3. Do I have to wait for the next release of pyadet to use this modification?